eclipse-threadx/guix #3: Build error for default example project in GUIX

xiongyu0523, 2020-05-27:
Latest VS 2019. GUIX Studio is installed. Double click a example project .sln under Build and start build. I got following error.
```
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\MSBuild\Microsoft\VC\v160\Microsoft.CppBuild.targets(448,5): warning MSB8003: The VCToolsInstallDir property is not defined. Some build tools may not be found.
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\MSBuild\Microsoft\VC\v160\Microsoft.CppBuild.targets(449,5): warning MSB8003: The WindowsSDKDir property is not defined. Some build tools may not be found.
1>TRACKER : error TRK0005: Failed to locate: "CL.exe". The system cannot find the file specified.
```

jdeere5220, 2020-05-27:
Hello xiongyu0523- Visual Studio contains (or doesn't include) many different compilers and tools. From your error message, it doesn't look like the C compiler has been installed. If you use Help|About, you should see a list of the installed components: 
[image]

xiongyu0523, 2020-05-28:
Appreciate for your help. 

I have this show up but I did not install Desktop C++ toolchain before. After install this component, everything is fine.

closed 2020-05-28 by xiongyu0523