eclipse-threadx/guix #4: Naming conflict with GNU GUIX

BonfaceKilz, 2020-05-28:
I don't know where else to post this; but hasn't the GUIX name already been taken by https://guix.gnu.org/. Coming from that ecosystem, it's abit confusing when I stumbled across this whilst trying to solve some guix(read gnu) related problems. Perhaps(though I bet highly unlikely but it doesn't hurt to try) a rename? You can close this anytime :shrug:

jdeere5220, 2020-05-29:
Unlikely to be able to change the name at this point. GUIX was actually initially release in 2012 I believe.

closed 2020-05-29 by jdeere5220

emixa-d, 2022-10-09:
I stumbled upon this issue, and noticed ...

> Unlikely to be able to change the name at this point. GUIX was actually initially release in 2012 I believe.

... that according to your (as in, this project's) own release notes (https://web.archive.org/*/https://github.com/azure-rtos/guix/releases/tag/v6.0_rel), the initial version is in 2020), not 2012.  Compares to GNU Guix, GNU Guix first v1.0.0 release was in 2019 (before Azure RTOS Guix!), according to 'git log'.

(There are also pre-1.0.0 releases from back in 2013, if you count v0.1!)

That was all I wanted to say.

BonfaceKilz, 2022-10-10:
emixa-d ***@***.***> anaandika:

> I stumbled upon this issue, and noticed ...
>
>
> ... that according to your (as in, this project's) own release notes [...]

+1 for pointing this out!

-- 
(Life is like a pencil that will surely run out,
    but will leave the beautiful writing of life.)
(D4F09EB110177E03C28E2FE1F5BBAE1E0392253F
    (hkp://keys.openpgp.org))

goldscott, 2022-10-10:
@emixa-d @BonfaceKilz - GUIX was actually released back in 2013, before Microsoft acquired Express Logic. Here's a link from wayback machine: https://web.archive.org/web/20131205113407/http://expresslogic.com/

When Microsoft acquired Express Logic, we started with fresh dates at version 6.0.

BonfaceKilz, 2022-10-11:
Scott Larson ***@***.***> aliandika:

> @emixa-d @BonfaceKilz - GUIX was actually released back in 2013, before Microsoft acquired Express Logic. Here's a link from wayback machine: https://web.archive.org/web/20131205113407/http://expresslogic.com/
>

Thanks for this clarification.  This issue's
thread will indeed serve as a reference point to
clear up things for the curious internet nomad.

> When Microsoft acquired Express Logic, we started with fresh dates at version 6.0.

Just my 2 cents, but I hope that this doesn't
escalate to a flame war :)

Both cases make sense.  And optimistically, I'd
say this was a[n] [un]fortunate coincidence
depending your perspective.

That said, this would be an interesting case study
wrt trademarks and what they imply legally.  Like
what would happen if these 2 projects shared the
same exact name?  IANAL thus only thing I can do
is speculate, so it's pointless to push this
thought any further.

Finally, I'm curious, how did you guys come to the
name "GUIX".  Does it have any semantic meaning
(say as an acronym) for something in the RTOS
space?  For GNU Guix, IIUC, it was a derivative of
"Nix" - a functional package manager /+ OS.  The
creators of Guix adopted GNU Guile as the
scripting language of choice.  So the "N" was
transformed into a "Gu" hence "Guix".

PS: You don't have to answer this if you don't
have the space/band-width to do so - no pressure.

-- 
(Life is like a pencil that will surely run out,
    but will leave the beautiful writing of life.)
(D4F09EB110177E03C28E2FE1F5BBAE1E0392253F
    (hkp://keys.openpgp.org))

goldscott, 2022-10-11:
@BonfaceKilz - all of the other Express Logic products had similar names (ThreadX, USBX, NetX, FileX), so it seemed only natural to name the GUI product "GUIX."