eclipse-threadx/guix #9: Reverse byte option for GUIX Studio

XuNeo, 2020-11-25:
Hi, 
I'm using SPI interface LCD with 16bit RGB565 pixel format. When 16bit data sent to LCD controller, the most efficient way is  send block of pixel data byte by byte using DMA. So the it would be better to have option to reverse the byte order of 16bit pixel data.

GUIX Studio has a checkbox "reverse byte order", however, it turns out to change RGB565 to BGR565. 
For source code, there are macros to assemble and dissemble color in gx_display.h. 
https://github.com/azure-rtos/guix/blob/89b07838c72c60a5d79e50f87ac2714284e634e3/common/inc/gx_display.h#L361-L371

It should be easy to change byte order, but I found the macro is not used in some of the c files, like 

https://github.com/azure-rtos/guix/blob/89b07838c72c60a5d79e50f87ac2714284e634e3/common/src/gx_display_driver_16bpp_pixelmap_rotate.c#L22-L31
and

https://github.com/azure-rtos/guix/blob/89b07838c72c60a5d79e50f87ac2714284e634e3/common/src/gx_utility_16bpp_pixelmap_resize.c#L22-L31

Is there plan to add "reverse byte order" to GUIX Studio, while not the RGB565 to BRG565 and fix the macro usage in some of the source files?

Thanks.

jdeere5220, 2020-11-25:
Hello XuNeo,

Yes, it's intentional that this option converts to BGR order, that's the hardware requirement that we have sometimes seen. To reverse the bytes as you describe sounds like more of a big-endian/little endian type change. Did you try turning on the BigEndian option?

I will enter a task to review the low-level functions and be sure they are using the ASSEMBECOLOR macro as they should. However I believe the bigger issue is the format of the pixelmap data produced by GUIX Studio. The ASSEMBLECOLOR macro isn't always used, especially if we are writing a solid color or rendering a glyph or pixelmap data with no alpha. In that case it's a straight write, and the source data would need to be in the desired format. This I believe requires a change on the GUIX Studio side.

I'm hoping that turning on the BigEndian option will give you want you want. Can you try that and report back please?

Best Regards,

Ken

XuNeo, 2020-11-26:
Hi Ken @jdeere5220 ,

Yes, I have tried the BigEndian option. But it won't work:

1. For compressed bitmap, it definitely won't work because data is not the original bitmap.
2. For non-compressed bitmap, it doesn't work and it seems like GUIX Studio's bug. The generated data won't change whether Big Endian box checked or not.

Ideally, GUIX Studio generates desired format of resources and source code checks data format and do conversion using macro.
I think to solve the 'byte order' issue, another color option like 3G_5B_5R_3G could be added and follow existing method dealing with BGR565. 3G_5B_5R_3G is just a byte-order reversed of RGB565, explanation below from LVGL.

```c

typedef union {
    struct {
#if LV_COLOR_16_SWAP == 0
        uint16_t blue : 5;
        uint16_t green : 6;
        uint16_t red : 5;
#else
        uint16_t green_h : 3;
        uint16_t red : 5;
        uint16_t blue : 5;
        uint16_t green_l : 3;
#endif
    } ch;
    uint16_t full;
} lv_color16_t;
```

Regards,
Neo

jdeere5220, 2020-11-29:
Hello XuNeo,

OK I will put this on our roadmap to support this format. I'm not sure how quickly we will be able to get to it, there are a few other things in front.

Best Regards,

Ken

yuxin-azrtos, 2021-01-28:
Close this issue.

closed 2021-01-28 by yuxin-azrtos