eclipse-threadx/guix #11: framelist of sprite is wrong when using GUIX studio 6.1.0.0

mwmy2009, 2020-12-15:
Hi

It might be a bug. When I create a sprite and define the frame list in GUX studio, the pixelmap ids shown in studio doesn't match those in the auto generated code. I hope you notice it and fix it in the next version.

BTW, about when the next version of GUIX studio is going to be released?

Best Regards
Wesley Fan

jltrahan-2020, 2020-12-22:
I second this bug...  It's been around since at least v5.5 or 5.6 of GUIX Studio (back before the MS acquisition.)  As Wesley reported above, GUIX Studio generates the wrong indices in the framelist.  The GX_SPRITE_FRAME array of frames have the wrong pixelmap id in the specifications.h file.  This shows in both the GUIX Studio interface, and the target compiled code.

The obvious workaround is to manually edit this .h file, but that's a pain in the long run, as it's not intended to be used this way.

jdeere5220, 2020-12-22:
Can you tell me what version of GUIX Studio you are using? I will look into this. We have several sprite based regression tests in place that are not exhibiting this issue so we will have to figure out what causes this to happen. The next GUIX Studio update is set for the end of this month, and we will do our best to duplicate this issue and resolve it for the next update. GUIX Studio is moving to the Microsoft App Store for the next release.

jltrahan-2020, 2020-12-22:
As the subject of the issue indicates, this is on the latest version 6.1.0.0.  I will send a private email with an example GUIX project.

jltrahan-2020, 2020-12-22:
@jdeere5220 Can you confirm that you received a private mail with a sample project?  I've even created another one from scratch using 6.1.0.0, thinking that importing an older .gxp file (generated by GUIX 5.5) would cause the issue, but even a fresh new project does not generate the correct pixelmap id's in the frame array.

jdeere5220, 2020-12-23:
@jltrahan-2020 Sorry I did not notice the thread title contained the version information. I did not receive a private email but I am looking into the issue. We will try to have it fixed for our end-of-this-month release.

jltrahan-2020, 2020-12-23:
Looking forward to test it.  Let me know if you need me to do so.

Jean-Luc

jltrahan-2020, 2020-12-23:
Let me know if you need someone to test the correction.  I'll take a look here once in a while during the Holidays.

Thanks!

jdeere5220, 2020-12-24:
@jltrahan-2020 we have reproduced the issue and fixed it (and added a new test to be sure it never comes back). The fix will be in our next Studio release, which I believe will be 6.1.3.0, early January.

jdeere5220, 2020-12-30:
@jltrahan-2020 just a side note, we did have a regression test in place for this code generation, but it turns out the test was in this case completely useless. The test project included only pixelmaps used in the sprite, and they were defined in alphabetical order, and used in alphabetical order, so of course the generated output was, in this case, correct. We have greatly improved the test going forward!

jltrahan-2020, 2021-01-04:
Thanks Ken, we look forward to using the update version!

yuxin-azrtos, 2021-01-29:
Hi @jltrahan-2020 we have released an updated GUIX Studio you can find at https://aka.ms/azrtos-guix-installer

Let us know if the newer version fixes the problem. Thanks!

jltrahan-2020, 2021-01-29:
Yes, the issue is fixed.

closed 2021-01-29 by yuxin-azrtos