eclipse-threadx/guix #12: Problem in executing using GUIX studio 6.1.0.0 in windows OS

armembedded, 2020-12-20:
I installed GUIX studio 6.1.0.0 in the windows operating system but the program does not start and display an error. 

how to resolve these issues?

error message title: 
The procedure entry point UiaDisconnectProvider could not be located in the dynamic link library UIAutomationCore.DLL

jdeere5220, 2020-12-23:
Can you provide your complete Windows version information please? Thank you!

jdeere5220, 2020-12-24:
I believe this error is caused by running an old version of Windows. You need to run minimum Windows 8, and starting next year Windows 10.

armembedded, 2020-12-26:
> Can you provide your complete Windows version information please? Thank you!

Windows 7 ultimate

jdeere5220, 2021-01-04:
Yes, you will need to update to Windows 10 to run GUIX Studio and TraceX applications.

yuxin-azrtos, 2021-01-28:
Looks like this issue is resolved.  Close it.  Feel free to reopen it if you have additional comments or feedback. Thanks!

closed 2021-01-28 by yuxin-azrtos