eclipse-threadx/guix #13: circular_gauge  Control error

geek-slt, 2020-12-28:
GUIX version: 6.1.0

jdeere5220, 2021-01-04:
This can be caused by using a compressed pixelmap for the needle image. The gauge performs runtime pixelmap rotation to rotate the gauge needle, and runtime rotation can only be done on uncompressed image. Right click on the image in GUIX Studio, Edit Settings, and make sure the "compress" option is turned off.

If that isn't the problem please get back to me and if possible attach your project.

geek-slt, 2021-01-05:
It seems that the needle can be displayed normally, but there are some residual images. I will continue to check. I am very interested in ThreadX and GUIX. Can I continue to ask you if I encounter problems later

jdeere5220, 2021-01-06:
Yes, of course, we are here to help. Send me your project and your images if you aren't able to resolve the problem so that I can dig into it. Another option is to compare your usage with our samples that use the circular gauge control.

yuxin-azrtos, 2021-01-28:
@geek-slt are you able to share your project with @jdeere5220?  Looking forward to your feedback.

geek-slt, 2021-01-31:
Ok 
Thank you very much, I will do as you say

jdeere5220, 2021-03-02:
No additional feedback, I will assume the suggestion fixed the issue reported.

closed 2021-03-02 by jdeere5220