eclipse-threadx/guix #16: GUIX Studio default pixelmap directory

jltrahan-2020, 2021-02-26:
Hi,

I recently had to re-install various things on my work PC, including GUIX Studio.  When I download 6.1.4.0 (the current version in the MS Store), it wants to install in C:\Azure_RTOS\GUIX-Studio-6.1 (note the "dashes".)  However, my existing GXP files actually point to C:\Azure_RTOS\GUIX_Studio_6.1 (note the "underscores".)  Was this default location changed at some point?  This breaks the builds of existing projects.

My underlying question: in GUIX Studio, why can't we delete the 4 default System pixelmaps that are not even used in the project, and completely avoid this issue?  RADIO_ON, RADIO_OFF, CHECKBOX_ON and CHECKBOX_OFF.

jdeere5220, 2021-02-26:
I'm a little bit confused by your comment. When you install GUIX Studio from the App Store, you really don't have an option where the app itself will be installed. The Windows installer takes care of that. Where previously the GUIX Studio installer copied example projects and default images to a directory of your choosing, now GUIX Studio prompts you to clone the GUIX git repo to your local machine. The default location for this clone is c:/Azure_RTOS. After the repo is cloned locally, you should see the default system images in the c:/Azure_RTOS/guix/graphics/system_png folder.  So yes things have changed quite a bit, but the situation you describe doesn't actually sound like the version released on the MS Store (I just installed this myself yesterday).

We do have a task in our backlog to eliminate the system png files and just include them internally like we do the default system fonts, we just haven't gotten to that task just yet. We do need the images though, just so that if you DO plop a radio button or checkbox on your screen we can show you *something*. You should be able to disable these default images so that they don't show up in your resource file.

jdeere5220, 2021-02-26:
@jltrahan-2020 could you please also provide the details of what is meant by "When I download 6.1.4.0"?  Do you mean you open the MS Store app, search for GUIIX, and click the "Get" button for GUIX Studio, or are you using some other method? I just want to clarify how you may have installed a rather odd trial version of GUIX Studio, if that is even the case.

jltrahan-2020, 2021-02-26:
That is correct, I used the MS Store, searched for GUIX, and then installed GUIX Studio that was found.  This installs 6.1.4.0.  Sorry, I used the term "download", but you are right, it was through the MS Store.

You are also correct about the installation directory, I don't have the option of choosing.  When I reinstalled GUIX Studio this morning, it installed the system png files under C:\Azure_RTOS\GUIX-Studio-6.1  Again, with "dashes".  Then, I proceeded to modify a GUIX .gxp project.  When I opened the .gxp file, GUIX Studio complained it couldn't find these png files.  Looking more closely, the .gxp file was telling GUIX to open the files in C:\Azure_RTOS\GUIX_Studio_6.1 (note, "underscores" here.)  I had to manually go in and modify the .gxp to point to the correct path for the png files.  I'm assuming a prior version of GUIX Studio I had (I forget which one, but it was in the 6.1.x series) was using the underscore and that now the dashes are being used to "locate" these png files.

Do this make more sense to you?

jdeere5220, 2021-02-26:
@jltrahan-2020 let me investigate and get back to you. I understand that the old "Studio Relative" path to the system png files needs to be updated, I'm sorry for that inconvenience. But the part about Azure_RTOS\GUIX-Studio-6.1 is still not clear, I don't know why you are having this path created. When you first run GUIX Studio, does it prompt you to automatically clone the GUIX git repo? The first time you run the new Studio installation, it should ask you where you want to put the git clone. Thanks for providing details.

jdeere5220, 2021-02-26:
@jltrahan-2020 actually if the git clone operation is working as expected, GUIX Studio will make registry entries to locate the new system_png folder. So you actually should not have to update any old project files. We do try hard not to break things. It still feels to me like the Store is not giving you the latest version, and I don't have much control over that unfortunately.

jltrahan-2020, 2021-02-26:
I just tested again.  I uninstalled GUIX Studio (accessible with right-click in the Start Menu.)  I also had to manually uninstall the Azure_RTOS directory as it's not removed automatically.  I then reinstalled GUIX Studio (using the MS Store.)  It installs 6.1.4.0.  When I first run it, it asks me if I want to Clone GUIX Repo, to which I click "I do not want to download samples at this time", as I don't need them.  It then still asks me where to install the examples, and that's where the "C:\Azure_RTOS\GUIX-Studio-6.1" is proposed.  This time around, I did notice that you could browse to another path here.  I assume for now that this is a standard path, and click continue instead, which then starts GUIX Studio.  If I open an existing project (created prior to 6.1.4.0), I then hit the "underscore" vs. "dash" naming problem for the 4 system PNG files.

A programmer here that would follow this process would likely also assume the "standard" location as well, as they don't know about this naming change.  This programmer then clones a project from Git to work on some new feature (or bug), and tries the "build all" script (which also regenerates the GUIX output files using the command line) before actually modifying any code.  At this point, the project no longer compiles because the PNG resources don't exist in the files (but are referenced in the generated code.)

```
/* Pixelmap Table                                                              */

GX_CONST GX_PIXELMAP *display_1_theme_1_pixelmap_table[] =
{
    GX_NULL,
    &DISPLAY_1_THEME_1_RADIO_ON_pixelmap,
    &DISPLAY_1_THEME_1_RADIO_OFF_pixelmap,
    &DISPLAY_1_THEME_1_CHECKBOX_ON_pixelmap,
    &DISPLAY_1_THEME_1_CHECKBOX_OFF_pixelmap,
etc...
```
It's not a critical problem, but it does create a bit of confusion, as when we clone from our Master branch in the project repo, we expect it to build right off.  As I mentioned earlier, being able to delete these 4 unused bitmaps would completely avoid the issue, as our code would be 100% "self contained", so to speak, and not depend on the location of files.  Imagine, in a near future, when you release GUIX Studio at version 6.2.0.0.  If you now install the default PNGs in C:\Azure_RTOS\GUIX-Studio-6.2, you would have the same problem when you open older .gxp projects created with GUIX Studio 6.1.4.0.

Again, it's not a big issue, but earlier today, it was confounding me (and another programmer who had last committed to our repo) until we realized the directory names were very similar, but not identical.

jdeere5220, 2021-02-27:
Thanks so much or your feedback. I understand now what is happening. I will look into what we can do to improve this experience.

jdeere5220, 2021-03-02:
@jltrahan-2020 We have added a task internalize these images so that no external path is needed. We will also confirm they can be removed from the application when not needed. Since we have added these tasks to our backlog I will close this issue, expect to see these changes in the near future.

closed 2021-03-02 by jdeere5220