eclipse-threadx/guix #17: Custom resource file for Fonts [Question]

StefanHengeleinLumira, 2021-03-11:
Our resource file is currently at 138MB (167MB with the latest version of GUIX studio) and we've seen an out of memory issue in our customers CI when compiling. So I was looking into splitting the file up to make it faster to compile and avoid issues in the CI.

I've found the following text in the current documentation (similar to the documentation of the GUIX version we're actually using):

> For Pixelmap and Font resources, you can specify a custom output filename for each pixelmap and font in theassociated resource editing dialogs. This feature allows you to put very large resources in distinct files, ratherthan putting all resources in one common output file. If you do not specify an overridden filename for a font orpixelmap resource, those resources are written into the common resource file.

I did indeed found a way to pull out the pixelmaps, however I wasn't able to find a way how to move the fonts to a custom file and the font specific documentation also doesn't mention this. Is this a removed feature? Or in the backlog? We'd really like to avoid writing a script to do the split up manually...Or am I just missing where/how this is done?

jdeere5220, 2021-03-11:
Hmmm... look like this was never implemented for fonts, I'm not sure why. I've added this task to the backlog. It may make it into the April update, but I can't promise that we have a pretty full task list for this update already.

ting-ms, 2021-04-12:
This feature has been added in GUIX Studio version 6.1.6.1.

yuxin-azrtos, 2021-04-21:
@StefanHengeleinLumira please the link https://aka.ms/azrtos-guix-installer to install the latest GUIX Studio.  Let us know if the new version addresses your issue. Thanks!

closed 2021-11-30 by jdeere5220