eclipse-threadx/guix #18: Partial Framebuffers

thilofrank, 2021-03-25:
Is GuiX able to handle Partial Framebuffers like TouchGFX?
Partial Framebuffers is a concept with at least two buffers. While one is transfered to the display, the others can be rendered and written. That works like a fifo and reduces the amount of needed RAM, because instead of a full screen framebuffer, two (or three) very small framebuffers are used.

jdeere5220, 2021-03-25:
Hello @thilofrank,

Support for this method of visible frame buffer update is in our task backlog and will be implemented for an upcoming update. Can you tell me, are you using  a self-refresh display with internal GRAM? Just curious.

thilofrank, 2021-03-25:
I use an ILI9341 spi display controller, that has GRAM included.

thilofrank, 2021-03-25:
The documentation mentions Ping-Pong-Framebuffers. Is this the same?

jdeere5220, 2021-03-25:
Regarding GRAM- perfect. Regarding Ping-Pong buffers, related but not exactly the same. GUIX currently supports full frame-buffer size ping-pong buffers, so the application is writing to and updating the working buffer while the LCD refresh circuitry operates out of the visible buffer. This is really intended for architectures without self-refresh, i.e. the MCU  or external LCD controller is continuously updating the LCD data. After each drawing sequence, the buffers are toggled. This is really implemented by the display driver, which also updates the GUIX canvas memory pointer after each buffer toggle.  

The partial buffer design is similar, but each working buffer is N scan lines in size. GUIX already implements full clipping, so the idea is that we clip the drawing to N scan lines, shift this out to the GRAM, draw the next N scan lines, etc... If you have two small buffers, drawing to buffer A and shifting buffer B to the GRAM can happen in parallel. GUIX also keeps track of the canvas dirty area, so often a full screen refresh is not required. The partial buffer design only redraws the dirty scan lines with each refresh operation. This requires a logic change in the internal GUIX canvas refresh function, so we can't implement this only in the hardware specific display driver. That is the reason a GUIX update release is required to support this correctly.

thilofrank, 2021-03-25:
The ILI9341 has a command to set its windows (x,y,w,h). If a partial framebuffer helds all data w/o the non-dirty areas, it can be flushed in one shoot. Is this your idea for the future?

azeater, 2022-06-22:
Just wondering if there's been any update on this? The idea of partial frame buffers is quite appealing in memory-constrained scenarios.

jdeere5220, 2022-06-22:
@azeater we have this task in our backlog but it's not going to make into the July update release. We should be able to tackle this for the next update release after July.

azeater, 2023-06-20:
Hi, sorry to revive a closed issue but I was wondering where this has been implemented/if it has been released yet?

ting-ms, 2023-06-20:
@azeater Apologies for closing this issue. The mentioned feature hasn't been supported yet, but we will consider supporting the feature in the future. If you are working with hardware from ST, Microchip, NXP, or Renesas, you may also seek assistance from them for further guidance and support on this matter.

ting-ms, 2023-08-31:
@thilofrank @azeater Is this feature still required? Could you provide information about the hardware you are utilizing and the specific scenarios in which you are employing this feature?

azeater, 2023-08-31:
Hi @ting-ms - yes I would definitely say this is required. Any embedded system with memory constraints would greatly benefit from this. For example, using a 320x240 16 bit display requires a full frame buffer of 153600 bytes - on a small MCU with 256KB of RAM that's over half the RAM needed to update the display which could be a blocker in using GUIX. 

@thilofrank mentioned TouchGFX but I've got experience with [LVGL](https://lvgl.io/) which also supports partial frame buffers and works quite well.

ting-ms, 2023-08-31:
@azeater Thank you for your response. Are you currently in the evaluation stage, or do you have a production waiting for the implementation of this feature?

azeater, 2023-08-31:
We are currently developing a product and are waiting on this feature.

ting-ms, 2023-10-31:
@thilofrank @azeater The partial frame buffer is currently available in GUIX release 6.3.0.
For detailed instructions on how to use this feature, please reference the documentation [Appendix D - Canvas Partial Frame Buffer Feature | Microsoft Learn](https://learn.microsoft.com/en-us/azure/rtos/guix/appendix-j).

closed 2023-10-31 by ting-ms