eclipse-threadx/guix #19: Circular gauge doesn't display correctly

Merc359, 2021-04-28:
Hi, 
I'm currently working on an interface that already has many windows (more than 30).
As soon as I add a cicular gauge widget (size 100px by 100 px) and change window and go back it colapse to a tiny little 2px by 2px.
Even though the data displays 100px by 100 px. This is purely through the use of the guix interface.

Thank you.
[image]

jdeere5220, 2021-04-28:
Hmmm that's interesting. I did a quick test to try and duplicate the issue, but so far it works fine. When I switch between window and window_1, nothing strange happens. I also tried this with samples/guix_widget_types.gxp, which includes a circular gauge window, and that also seems to come back fine. I can't see all of your gauge properties in your screen shot, can you compare your gauge to the gauge(s) in the guix_widget_types.gxp sample and detect what might be different? I will keep poking around but we need to reproduce what you are seeing. Thanks!

[image]

jdeere5220, 2021-04-28:
@Merc359 tag you to be sure you see reply above.

Merc359, 2021-04-29:
Hi,

Here is the problem in video.
Thanks

https://user-images.githubusercontent.com/83370744/116564187-be0c2e80-a8d2-11eb-8292-ac42e9bcaac9.mp4

jdeere5220, 2021-04-29:
@Merc359  I was able to reproduce the problem. We will fix it asap and push a new Studio release. For a work-around, I believe if you assign a background image to the gauge this will not happen. The issue is caused by the circular gauge attempting to auto-resize to fit the gauge background image. If there is no background image the gauge becomes very small.

Merc359, 2021-04-29:
That is good news. I will assign a background like you've suggested until the fix is released.
Thank you for the quick response and have a good day.

closed 2021-11-30 by jdeere5220