eclipse-threadx/guix #21: How drawing gradient color

RickMenza, 2021-06-07:
Hi,
I want to draw a slider using gradient color (black to a color returning by the application).
Since I don't know which color will be used by the application, I need to creat it at runtime.

I think I should use the gx_utility_gradient_create function but I don't know how to use it and draw a pixelmap containing my gradient color.

Is someone has a code snippet exemple to help me how to do?
Thanks in advance!
Richard

jdeere5220, 2021-06-07:
Hello @RickMenza ,

I whipped up a quick example. A couple of notes:

1) GUIX only supports creating ALPHAMAP gradients today, which means it generates an 8bpp alphamap and you can then runtime assign the color used to draw the alphamap by initializing your brush. That's probably what you want anyway (that's what everybody wants!) but just to make sure you understand what's happening.

2) I purposely made it so the code calls "CreateGradient" every time you push the button, even though it really doesn't need to because the same gradient is being used every time, just drawn in different brush colors. I just wanted to demonstrate that gx_utility_gradient_create always check to see if the requested gradient has already been created, and if so it returns the existing gradient alphamap instead of creating a brand new one. If you were to check on the amount of the scratchpad memory that is free, it should only decrease on the first call and then stay constant.

There is a VisualStudio 2019 project included that should allow you to build and run as a desktop app. Let me know if you have any issues.

Best Regards

[guix_gradient_test.zip](https://github.com/azure-rtos/guix/files/6611686/guix_gradient_test.zip)

closed 2021-06-07 by jdeere5220

RickMenza, 2021-06-08:
Hi Ken,

Thank you very much. Your example help me to understand the gradient functionality and I am now able to do what I want in my application.

Have a nice day!
Regards,
Richard