eclipse-threadx/guix #23: Lib file for GUIX Studio version 6

CYG01571, 2021-07-15:
Hi

I'm developing a GUIX application on Visual Studio 2019 Professional(Windows10).
I used to use the library files deployed in the folder "C:\Express_Logic\GUIX_Studio_5.6\win32_runtime".
I would like to upgrade the GUIX studio to the version 6 Azure RTOS GUIX Studio.
Could you provide these library files for version 6?
ex. gx.lib, gx_api.h, gx_port.h, gx_user.h, tx.lib, tx_api.h, tx_port.h

Best regards.

jdeere5220, 2021-07-15:
When you install the latest GUIX Studio from the Microsoft Store, it will prompt you to clone the GUIX source code repo from github to your local machine. The files you are looking for are included in the GUIX repo under /ports/win32. You can build gx.lib yourself using ports/win32/guix.sln, and the threadx lib and headers are pre-built for you in ports/win32/lib/vs_2019 folder. All of the examples are now set up to build using this structure so you should be able to open any of the projects under /samples and build/run on Win32 using VS 2019. Let me know if you have any issues.

closed 2021-07-15 by jdeere5220

CYG01571, 2021-07-16:
Thank you for your kind reply!
I'll try it.