eclipse-threadx/guix #24: Set display orientation during running

lixinlv, 2021-08-13:
Hi,
In my project, it needs to set the display oratation of LCD according to the placement position during operation, sometimes 90 degrees(clockwise or counterclockwise) and sometimes 180 degrees.
Previously used GUI_SetOrientation() in emWin is implemented. No similar function is found in GUIX. Can it be added in GUIX later versions?
Thanks.

jdeere5220, 2021-08-13:
I will add this our our backlog. However, be advised that in order to implement 90/180/270 degree rotation with no runtime overhead, we utilize GUIX Studio to produce the resource data in the rotated format. So supporting this application requires configuring each rotation as a different display in the GUIX Studio project, and producing a resource file and display instance for each rotation angle. So at runtime, you would actually switch your display instance (resources are linked to a display). Does that help?

lixinlv, 2021-08-14:
Hi jeddre5220,
Thanks for your reply.
Designing an respective instance for 0/90/180/270 degrees is a solution, but code will be increased. If the 0/180 degree interface can share the same resource and instance, code will be greatly reduced. So is 90/270 degrees. If API can provide rotation function, GUIX studio can design two interface instances for 0 and 90 degrees, in this way code can be half reduced at least. What do you think about this suggestion?
Thanks.

jdeere5220, 2021-11-30:
The way our code generation for rotated display works today, we want to be able to fully utilize any underlying hardware accelleration for 0/90/180/270 rotation, so we generate the resource data such that it can be streamed out in all rotation angles just like it is done for no rotation. I have added this suggestion to our backlog but I can't say how quickly we can investigate further.

closed 2021-11-30 by jdeere5220

lixinlv, 2021-12-01:
Hi jeddre5220,
Thanks for your reply.
I'm looking forward to your further upgrade.