eclipse-threadx/guix #27: Menu Widget Text Alignment

brhans-acuity, 2021-11-08:
The 'Text Align' attribute applied to a 'menu' widget in GUIX Studio is not respected in the generated code.
In my project, I have a 'tree_view' which in turn contains a number of 'menu' widgets.
For each 'menu' widget I have set the 'Text Align' attribute as 'Center', and the text is displayed correctly center-aligned in GUIX Studio.

When I save the GUIX .gxp project and inspect the XML code, I see that the <style> value is what I expect it to be - in this case 1610661888 (or 0x6000C000), indicating to me that "GX_STYLE_TEXT_CENTER 0x00004000UL" has been saved.

However, when I generate the C code for my project, the style flags are missing GX_STYLE_TEXT_CENTER - only "GX_STYLE_BORDER_NONE | GX_STYLE_DRAW_SELECTED | GX_STYLE_ENABLED | GX_STYLE_TEXT_COPY" is present, and when running the compiled code on my hardware the text is left-aligned instead of center-aligned.
If I manually edit the generated code to add GX_STYLE_TEXT_CENTER to the style for my 'menu' widgets, then the test is center-aligned as expected.

jdeere5220, 2021-11-09:
@brhans-acuity thank you for the very thorough description! Based on your description alone, I can pretty confidently say this is a bug (sorry to restate the obvious). I will enter a bug in our backlog right away and get this fix into our next patch release. I will try to notify you when that is posted, but if I forget and you don't hear back please ping me. I would expect we can have this update posted within a couple of weeks if not sooner.

brhans-acuity, 2021-11-09:
Hi Ken,
Thanks for the speedy response.
Is there somewhere ‘official’ I could submit feature requests for GUIX Studio?
I’ve spent a fair amount of time recently using other GUI creation tools (mostly Embedded Wizard - https://www.embedded-wizard.de/), and I feel like there are some features which would really make GUIX more “powerful”.
My biggest request would be around making it easier to re-use component widgets created as templates in different projects – to be able to ‘export’ a component widget I’ve created and ‘import’ it into a different project.
I’m currently doing this manually by performing cut-and-paste of he XML between the GUIX gxp project files, but it’s really hacky and prone to introducing bugs (I’ve managed to put GUIX Studio into an endless loop a couple of times by having a widget reference itself as a template …)

Bruce Hansen
Principal Design Engineer

Acuity Brands Lighting, Inc.
1 Acuity Way |  Decatur/GA/30035
o 770 860 2835
www.acuitybrands.com<https://urldefense.proofpoint.com/v2/url?u=https-3A__acuity.pmail4.com_PoliteMail_default.aspx-3Fpage-3DuARy1b38okaVlQOSDZZV7g-26ref-5Fid-3DvvBE4AV-5FwUONh9xbeovEZw&d=DwMFaQ&c=tvvHx3uC0XdtgG-ImPDjfM0QnakIsmmGcwejQDVxD-g&r=XvlYhLo2iDEBJO0vEeL5dUZ3rKph30LTfYAthxtknHg&m=0ti2Eg6Dtl-D7yY15c1ovd0xXL51xvn2KKZudT2cXxk&s=2HoyM-zOq8zGXaqWBhD2xgbEKpHvd7IyLMbHcCWRRCs&e=>
***@***.***<https://urldefense.proofpoint.com/v2/url?u=https-3A__acuity.pmail4.com_PoliteMail_default.aspx-3Fpage-3DN1bfPpkX3E-2DADFi9srpZ1w-26ref-5Fid-3DvvBE4AV-5FwUONh9xbeovEZw&d=DwMFaQ&c=tvvHx3uC0XdtgG-ImPDjfM0QnakIsmmGcwejQDVxD-g&r=XvlYhLo2iDEBJO0vEeL5dUZ3rKph30LTfYAthxtknHg&m=0ti2Eg6Dtl-D7yY15c1ovd0xXL51xvn2KKZudT2cXxk&s=taPrh4H2OjzUwjGBVHmluZleA2AMxDxGdAlst2zuYXI&e=> ***@***.*** <https://urldefense.proofpoint.com/v2/url?u=https-3A__acuity.pmail4.com_PoliteMail_default.aspx-3Fpage-3DNVjBBoBo-2DUCFgfcFarFhZA-26ref-5Fid-3DvvBE4AV-5FwUONh9xbeovEZw&d=DwMFaQ&c=tvvHx3uC0XdtgG-ImPDjfM0QnakIsmmGcwejQDVxD-g&r=XvlYhLo2iDEBJO0vEeL5dUZ3rKph30LTfYAthxtknHg&m=0ti2Eg6Dtl-D7yY15c1ovd0xXL51xvn2KKZudT2cXxk&s=wUkuqwZC9PQfUX644UMqk53XoBuqaR1WNb7nd01yxkI&e=>  ***@***.*** <https://urldefense.proofpoint.com/v2/url?u=https-3A__acuity.pmail4.com_PoliteMail_default.aspx-3Fpage-3DPioekOQKMkSM4anyrtifsw-26ref-5Fid-3DvvBE4AV-5FwUONh9xbeovEZw&d=DwMFaQ&c=tvvHx3uC0XdtgG-ImPDjfM0QnakIsmmGcwejQDVxD-g&r=XvlYhLo2iDEBJO0vEeL5dUZ3rKph30LTfYAthxtknHg&m=0ti2Eg6Dtl-D7yY15c1ovd0xXL51xvn2KKZudT2cXxk&s=dWykJS9XGbHq2jrJqW8tdy48dkLtyAvDgoZipr8u5e0&e=>  ***@***.*** <https://urldefense.proofpoint.com/v2/url?u=https-3A__acuity.pmail4.com_PoliteMail_default.aspx-3Fpage-3Do8YkH67pZkivLjaeui2NpA-26ref-5Fid-3DvvBE4AV-5FwUONh9xbeovEZw&d=DwMFaQ&c=tvvHx3uC0XdtgG-ImPDjfM0QnakIsmmGcwejQDVxD-g&r=XvlYhLo2iDEBJO0vEeL5dUZ3rKph30LTfYAthxtknHg&m=0ti2Eg6Dtl-D7yY15c1ovd0xXL51xvn2KKZudT2cXxk&s=FS4Fc3Ot7K5wxwdPC89ezJmLbOXBBWAKbDAZF6CA7-s&e=>  ***@***.*** <https://urldefense.proofpoint.com/v2/url?u=https-3A__acuity.pmail4.com_PoliteMail_default.aspx-3Fpage-3DzaT5bYceq0-2D-2DmV8-5FyJt3qg-26ref-5Fid-3DvvBE4AV-5FwUONh9xbeovEZw&d=DwMFaQ&c=tvvHx3uC0XdtgG-ImPDjfM0QnakIsmmGcwejQDVxD-g&r=XvlYhLo2iDEBJO0vEeL5dUZ3rKph30LTfYAthxtknHg&m=0ti2Eg6Dtl-D7yY15c1ovd0xXL51xvn2KKZudT2cXxk&s=e12JlApov1SlHWw5NzUV9rPpPpzDrRy7dWBfR0gs6WY&e=>     Subscribe to our email updates<http://www2.acuitybrands.com/subscription-center>
________________________________________________________________________
Aculux  |  American Electric Lighting  |  Antique Street Lamps  |  Atrius  |  DGLogik
Distech Controls  | DTL  |  eldoLED  |  Gotham  |  Healthcare Lighting  |  Holophane
Hydrel  |  Indy  |  IOTA  |  Juno  |  Lithonia Lighting  |  Lucid  |  Mark Architectural Lighting
nLight  |  Peerless  |  RELOC Wiring  |  ROAM  |  Sensor Switch  |  Sunoptics  |  Winona

jdeere5220, 2021-11-09:
Hello @brhans-acuity , 

As far as submitting feature requests, we are a pretty small team and we are all notified of queries made here. When someone requests a reasonable addition I add it to the "customer requests" Feature of our task backlog in DevOps, so it's not lost as long as I do that step! The hardware question is when we will a particular ask completed. Bug fixes take priority, but simple additions can be added pretty quickly as well.

Regarding copying from one project to another, you should be able to simply select anything you want to copy in one project and then open a second instance of GUIX Studio and paste that selection. We did a lot of work to support that, and it should even import all of the resource dependencies, templates, and such needed to support whatever you copied/pasted. Of course if the resources aren't actually there in the same project-relative or absolute path locations you will have to fix that up after the "paste", but it is supported.

There is also an "import" command under the Project menu. If you click on any folder in the Project view this command should be enabled, it allows you to open a second project (without starting a second instance of Studio) and import what you want from that second project into your current project.

Let me know if those options aren't providing the capability for which you are searching. I have already created a bug in our backlog regarding the text-alignment issue so it will get attention.

Best Regards,

jdeere5220, 2021-11-30:
Fixed for 6.1.9.0 release.

closed 2021-11-30 by jdeere5220