eclipse-threadx/guix #31: We want to install GUIX Studio locally

CYG01571, 2021-12-08:
Hi!

We have used GUIX Studio on Windows10 at company.
We can't download from the Store app due to company IT policy. 
Previously, Windows installer was prepared, could you prepare a package that can be installed locally?

bo-ms, 2021-12-08:
@yuxin-azrtos any comments on this issue? thanks.

yuxin-azrtos, 2021-12-08:
@CYG01571 please contact azure-rtos-support@microsoft.com and the support team can help you.

CYG01571, 2021-12-14:
Thank you, @jdeere5220 .

The latest version of GUIX Studio is V6.1.7.0 at Microsoft Store.
Renesas provides GUIX Studio software at their site, the version is V5.6.1.0.
And they declare that they don't provide the latest version software any more, download at Store...
(https://www.renesas.com/jp/en/software-tool/guix-studio)
 So I have questioned at this site...
 
 Best Regards.

jdeere5220, 2021-12-15:
@CYG01571 thanks for the feedback. To clarify, the latest version we have posted to the Microsoft App Store is version 6.1.9.2. This is a minor patch/bug fix to correct a reported error. We have provided the standalone installer link to Renesas for every version up to 6.1.9.1.  It is my understanding that we want to encourage developers to use the App Store version for security reasons. We are able to "push" updates via the Store channel in the event that some security threat is discovered. We can't do that with the standalone installer. In any case, you can (for now) download the standalone installer using this link: https://expresslogic.sharefile.com/d-s9055403587414096bfe3e70b4ce815e8  

Let me now if you have any issues.

closed 2021-12-15 by jdeere5220

CYG01571, 2021-12-16:
Thank you for your effort, @jdeere5220 !

OK. I'll try it.
I try to send e-mail to microsoft.

Best Regard