eclipse-threadx/guix #36: Canvas uses dynamic memory, no malloc memory for canvas?

HelloByeAll, 2022-04-26:
Now I don't have it checked in Studio ` alloccate canvas memory `，

gx_canvas_create.c -> _gx_canvas_create(...)
It works fine after adding the following code on line 113

```
    else
    {
        canvas->gx_canvas_memory = (GX_COLOR*)_gx_system_memory_allocator(memory_size);
        memset(canvas->gx_canvas_memory, 0, memory_size);
    } 
```

jdeere5220, 2022-04-26:
I feel I might be missing part of the question or the history, but I can comment on what I see. You are 100% correct, GUIX does not automatically invoke your dynamic memory allocator to reserve the canvas memory. Embedded systems are diverse with often very precise needs for where things are placed, especially large things like canvas memory buffers. Often various hardware requires the canvas memory be aligned on a certain modulo boundry, or very often the system will have a small, local, fast memory area used for stack and dynamics, and a slower external SDRAM or similar that is used for the canvas memory. Very often the low-level display driver itself reserves the canvas memory in a way to meet the specific requirements of that particular architecture.  For example, when running on Renesas Synergy targets, the canvas memory buffer(s) are reserved using the Renesas system configurator tool. Bottom line we leave it to the application code to decide where the canvas memory should be located, and what you are doing above is exactly correct (although it's maybe a bit unusual to use your dynamic memory pool for this, it's not wrong).  So your application along with your display driver need to decide where the canvas memory will be located, and we provide an API gx_canvas_memory_define() that is a little bit cleaner way to assign your canvas memory pointer. Hope this helps.

HelloByeAll, 2022-04-26:
Thank you for your answerThank you for your answer

closed 2022-04-26 by HelloByeAll