eclipse-threadx/guix #38: how to porting the guix to freertos?

Barfell, 2022-06-22:
there has a question, some guix wiget need dynamic memeory, need the tx_byte_pool_create() function, but the function in the threadx.
how to fix it?

	/* create a byte pool to use for rotating the needle bitmap */
	tx_byte_pool_create(&memory_pool, "guix", guix_pool_memory, 1024);

	/* install our memory allocation / de-allocation routines */
	gx_system_memory_allocator_set(memory_allocate, memory_free);

Barfell, 2022-06-22:
I'm defined a text prompt,and a timer,in the timer int, refresh the text. the text not change.
but ,test guix + theadx,the text refreshed. because used the function tx_byte_pool_create();

UINT _cbEventWindow_language(GX_WINDOW *widget, GX_EVENT *event_ptr)
{
    static uint32_t count = 6;
    char buf[1] = {0};

    
    switch (event_ptr->gx_event_type)
    {
        /* 控件显示事件 */
        case GX_EVENT_SHOW:
          
            /* 启动一个GUIX定时器 */
            gx_system_timer_start((GX_WIDGET *)widget, GUI_ID_Timer0, 1, GX_TICKS_SECOND);

            gx_window_event_process(widget, event_ptr);
            break;
        case GX_EVENT_TIMER:
            if (event_ptr->gx_event_payload.gx_event_timer_id == GUI_ID_Timer0)
            {
                sprintf(buf, "%d", count--);
                gx_prompt_text_set((GX_PROMPT *)&(window_language.window_language_prompt_countdown), buf);
            }
            break;

        default:
            return gx_window_event_process(widget, event_ptr);
    }

    return 0;
}

jdeere5220, 2022-06-22:
For the byte pool question, GUIX basically leaves it up to the application to define how the thread-safe heap works. If you want to use features that require dynamic memory, you need to implement a memory alloc/free function set using FreeRTOS. I am not familiar enough with FreeRTOS to provide you an example, but they must have functions similar to the ThreadX byte pool functions.

Barfell, 2022-06-22:
thank a lot ,the timer event happening, the memory alloc/free function need Implement。
I`m use freertos + st emwin, in the emwin soft timer event, refresh the text, but，dozens of days，the text not refresh, can not online debug, so ,the B plan is use freertos + guix, only need to change the graphics driver。

jdeere5220, 2022-06-22:
Let me know if you have any questions about the memory alloc/free implementation. The private text copy prompt style also requires a dynamic memory pool.

Barfell, 2022-06-22:
I do not know hao to implement the alloc/free function in guix, it looks a bit complicated. very amateurish in programming. In most cases, use middleware。

Barfell, 2022-06-22:
the threadx alloc/free function ,looks use linked list.

Barfell, 2022-06-22:
It is not clear, never studied the FREE RTOS kernel implementation in depth, usually use the encapsulated functions directly, should have, the memory management service,I`m  try to port it.

flyflyfly001, 2022-06-27:
Hi Barfell, you can reference this memory allocate/free function

VOID* memory_allocate(ULONG size)
{
    return pvPortMalloc((uint32_t)size);
}

VOID memory_free(VOID* mem)
{
    vPortFree(mem);
}

Barfell, 2022-06-28:
I'm try to port the code, I found the tx_byte_allocate can specify the size.

UCHAR guix_pool_memory[64];

/*******************************************************************************/
VOID *memory_alloc(ULONG size)
{
    VOID *memptr = NULL;

		memptr = /*(MEM_BLOCK_t *)*/osMemoryPoolAlloc(guix_pool, 0U);
		if(memptr != NULL)	
//    if (tx_byte_allocate(&guix_pool, &memptr, size, TX_NO_WAIT) == GX_SUCCESS)
    {
        return memptr;
    }
    return GX_NULL;
}

/*******************************************************************************/
VOID memory_free(VOID *memptr)
{
 //   tx_byte_release(memptr);
	osMemoryPoolFree(guix_pool, memptr);
}

	guix_pool = osMemoryPoolNew((uint32_t)16, sizeof guix_pool_memory, NULL);
	
	/* install our memory allocation / de-allocation routines */
	gx_system_memory_allocator_set(memory_alloc, memory_free);

closed 2022-12-20 by jdeere5220