eclipse-threadx/guix #39: Is it possible to use GUIX Studio export font resource as a bin file seperately ?

flyflyfly001, 2022-06-25:
Hi , we already know that GUIX Studio is able to export all resource data as a bin file. However, font resource are usually very large, which is not efficient when doing OTA with BLE. In most situlation, pixelmap and strings may need to upgrade while font is not. It is very friendly if we can load font resource and other resource seperately. Any reply would be much appreciated.

jdeere5220, 2022-06-27:
You should have full control over what is placed in the generated resource file, both when using GUIX Studio UI environment and from the command line. If I am missing something from your question please follow up.

int-bio, 2022-06-27:
We have done this in order to move the fonts and bitmaps to external flash.  I cannot give you the script, but I can give you a hint as to what you must do.  

============================================================
1.	Understanding the Main Tables
    GUIX is designed to support multiple themes.  A Theme describes how the screens 
    will look to the user.  Our project currently supports one theme.  Colors, Fonts, 
    and pixel-maps are defined in the Display Theme table.

    GX_THEME display_1_theme_1 =
    {
        (GX_COLOR *) display_1_theme_1_color_table,
        (GX_FONT **) display_1_theme_1_font_table,
        (GX_PIXELMAP **) display_1_theme_1_pixelmap_table,
      ...
    }
    The language table is defined in a separate table.
    GX_STUDIO_DISPLAY_INFO gui_display_table[1] =
    {
      ...
        display_1_language_table,
        DISPLAY_1_THEME_TABLE_SIZE,
      ...
    };

============================================================
2.	Splitting up resources
    The gui_resources.c file needs to be separated into multiple files so that 
    we can place the objects into other sections of memory:
    1)	guix_language.c 
        a.	Section .language_ro
    2)	guix_bitmap.c 
        a.	Section .bitmap_ro
    3)	guix_font.c 
        a.	Section .font_ro

    The color table is too small to consider moving to QSPI.

============================================================
3.	File updates
3.1. gui_utils.h
    /// @brief Copies LANGUAGE QSPI data to SDRAM
    void update_language_table(void);

    /// @brief Copies BITMAP QSPI data to SDRAM
    void update_bitmap_table(void);

    /// @brief Copies FONT QSPI data to SDRAM
    void update_font_table(void);

3.2. init_thread_entry.c

    // Initializing the i2c mutex early.
    i2c_mutex_init();

    // Copy QSPI data to SDRAM
    update_language_table();
    update_bitmap_table();
    update_font_table();

    // call function in images
    init_function();

4.	Fonts
Fonts will be moved to QSPI space. A duplicate font space will be created in SDRAM.  
Upon Boot, the QSPI data will be copied to SDRAM.

4.1. guix_font.c
    1)	Scan each line of file for  “static GX_CONST GX_UBYTE”
    2)	Allocate SDRAM space for font
        a.	static GX_UBYTE FONT_THEME_1_PROMPT_char_2026[24] BSP_PLACE_IN_SECTION_V2(".sdram");
        b.	no GX_CONST required for .sdram
    3)	Move font to QPSI
        a.	static GX_CONST GX_UBYTE FONT_THEME_1_PROMPT_char_2026_ro[24] BSP_PLACE_IN_SECTION_V2(".font_ro") =
    4)	Add font name to list
    5)	When no more lines are found, create the function update_font_table().

    // add function to bottom of file.
    void update_font_table(void)
    {
        memcpy(FONT_THEME_1_PROMPT_char_2026,  FONT_THEME_1_PROMPT_char_2026_ro, sizeof(FONT_THEME_1_PROMPT_char_2026_ro));
    }
    
4.2. S7G2.icf
    Add .font_ro section to arch\arm\ib\ngt\s7\script\S7G2.icf

    do not initialize   { section .qspi_flash };
    do not initialize   { section .font_ro };
    do not initialize   { section .qspi_flash_ro };

    place at start of QSPI_region      { block IB_QSPI };
    place in QSPI_region      { section .font_ro};
    place in QSPI_region      { section .qspi_flash_ro};

5.	Language
    By moving all strings and tables to SDRAM, we only have to update the display_1_language_table.  
    Initially, the linker will point all string tables to QSPI.  During boot, code will copy the 
    language strings and tables from QSPI to SDRAM and update the pointers in display_1_language_table to point to SDRAM.

    GX_CONST GX_UBYTE display_1_DEGREE_C_English[4] = {0xc2, 0xb0, 0x43, 0x00};
    GX_CONST GX_UBYTE display_1_OK_English[3] = "OK";

    GX_CONST GX_UBYTE *display_1_English_string_table[351] =
    {
        GX_NULL,
        display_1_DEGREE_C_English,
        display_1_OK_English,	
        ...
    }

    GX_CONST GX_UBYTE **display_1_language_table[12] = 
    {
        display_1_English_string_table,
        display_1_French_string_table,
        display_1_German_string_table,
        display_1_Spanish_string_table,
        display_1_Swedish_string_table,
        display_1_Catalan_string_table,
        display_1_Dutch_string_table,
        display_1_Danish_string_table,
        display_1_Norwegian_string_table,
        display_1_Finnish_string_table,
        display_1_Italian_string_table,
        display_1_Portuguese_string_table,
    };

    The strings must be located in SDRAM; the tables can be located in either SDRAM or RAM.

flyflyfly001, 2022-07-01:
@int-bio Hi, int-bio, thanks for your reply. I want to use XIP mode instead of load resource to SDRAM. I'm studying the GUIX source code. If there is any result, I will share with you.

flyflyfly001, 2022-07-19:
> You should have full control over what is placed in the generated resource file, both when using GUIX Studio UI environment and from the command line. If I am missing something from your question please follow up.

Yes, that's correct. We can generate fond resource as a bin file, and use **gx_display_font_table_set** to load font table to current display. Thanks a lot.

closed 2022-07-19 by flyflyfly001