eclipse-threadx/guix #42: Can generic scroll sheel be used as a horizontal scroll wheel ?

flyflyfly001, 2022-08-08:
Hi, there
      We want to use a horizontal scroll wheel. But the default scroll wheel that GUIX have is in vertical direction. Can generic scroll sheel be used as a horizontal scroll wheel ? Or is there any much better way to realize a horizontal scroll wheel ? Any help would be much appreciated.

jdeere5220, 2022-08-08:
Unfortunately our scroll wheel widgets are designed for vertical orientation. Event processing, child positioning, wheel wrapping, rows, row-height, etc.. all designed for a wheel that scrolls up and down.

That said, we are always on the lookout for new widgets or adding new styles to existing widgets. I will look into it and see if it makes sense to add support for horizontal orientation to the existing widget or if we should implement an entirely new widget type.

If you are ambitious you can implement your own horizontal scroll wheel widget type, but that would be quite a bit of work and you wouldn't have support in GUIX Studio. I mention it only if you are under time constraints.

flyflyfly001, 2022-08-09:
> Hi, jdeere5220
   Thanks for you reply. I just read the generic scroll wheel source code. The logic can be shared with horizontal scroll wheel except row and row height handling. I will try to realize it by myself.

closed 2022-08-10 by flyflyfly001