eclipse-threadx/guix #44: 为什么会进Hardfalut

LiuChenghu, 2022-08-11:
[image]
在如图所示的地方进Hardfalut,这什么原因导致的呢。

jdeere5220, 2022-08-11:
I'm going to need quite a bit more information here. Can you tell me what the application is doing that leads to this? Has the child widget somehow been attached to multiple parents?

LiuChenghu, 2022-08-11:
使用的是ThreadX + GuiX，做的是一个小弹窗，弹出一个小的界面，这个现象不是每次都会出现进入Hardfault,而是偶尔出现一次。
[image]从图中进入，小弹窗，下面是这个图片是弹窗的实现
[image]
显示的静态内容如下
[image]

jdeere5220, 2022-08-11:
I don't like that last line:

**g_pPopPartTimeScreen->gx_widget_parent = parent;**

You shouldn't directly assign the gx_widget_parent member. There could be another parent that already has g_pPopPartTimeScreen as a child, which would lead to the issue you originally reported. You should use an API, such as 

gx_widget_attach(parent, g_pPopPartTimeScreen); 

instead. That will automatically detach the g_pPopPartTimeScreen if it was previously attached somewhere else. Let me know if this fixes the problem.

closed 2022-12-20 by jdeere5220