eclipse-threadx/guix #46: gx_binres_theme_load consumed to large RAM

flyflyfly001, 2022-09-19:
Hi there,
     I'm using GUIX Studio generated font resource as a binary file. The binary file is about 10MB. I put the bin file to external qspi flash and maped the flash to MCU with XIP mode. When I use gx_binres_theme_load function to load the font table. The memory allocate is failed. Because my MCU has only 600KB ram. But this function try to malloc 1572KB memory. Why this function need to much memory instead of get font data directly form bin data ? Any help would be much appreciated.

jdeere5220, 2022-09-19:
Hello @flyflyfly001,

For fonts, there are two pieces of data for each glyph. First is the GX_GLYPH structure, which contains details about the glyph width, height, baseline, advance, and a pointer to the glyph data. The second piece is the actual glyph rendering data, which (depending on format) is an alphamap that determines the glyph appearance.

The way the binary data load works today, we create an array of GX_GLYPH structures in local RAM, and initialize those GX_GLYPH structures to point to the glyph data located in the binary data blob. I imagine the large memory allocation is mostly for these local GX_GLYPH array(s), but we would need to step through the _gx_binres_theme_buffer_allocate() function to be 100% sure about that.

There are a few reasons it works this way, but mainly the GX_GLYPH data in the binary data blob doesn't (and cannot) actually contain a pointer to the glyph data. It contains an offset, which we convert to a pointer when we read and initialize the local copy of the GX_GLYPH structures.

We are open to changing how this works if we can. Just thinking out loud, if the GX_GLYPHS table pointed to by the GX_FONT structure points to glyph data containing offsets, not pointers, then a few things would need to change outside of the binary resource loading logic. Today after a binary resource is loaded the rest of the library and GX_CANVAS functions don't need to know where the data came from. If we were to make a change such that GX_GLYPH structures might contain an offset rather than a pointer to the glyph data, it means the glyph rendering logic would need to be able to calculate an address using the binary data base address and the offset contained in the GLYPH structure. It gets a bit ugly because the binary data base address isn't something we keep around today after the load operation is complete.

Maybe we can start by verifying the reason for the large memory allocation (I suspect you have a lot of glyphs) by stepping through _gx_binres_theme_buffer_allocate(), and then we can look further into options for resolving this.

Best Regards,

flyflyfly001, 2022-09-26:
Hi @jdeere5220 
    I stepping through _gx_binres_theme_buffer_allocate() and make sure that it allocate large memory. Because I need to use a lot of _Kanji Ideographs_ characters. This will take a lot of storage space. And at the same time, may need to consume large memory to store GX_GLYPH. 
    I have found another way to solve this problem. I generate font resource as a source file instead of binary file. And tell the compiler to link font_resource.c at a fixed address which is located at external nor flash. This works well for me.
   Thanks for you kind help !  -_-

closed 2022-09-26 by flyflyfly001