eclipse-threadx/guix #49: Any case porting port_smp/linux/gnu causes a deadlock

jiale-gdyd, 2022-10-16:
[DRMMEDIA][I]: Enable vo channel:[1] starting......
[DRMMEDIA][D]: DrmDisplay flow params:[name=drm_output_stream
 device=/dev/dri/card0
framerate=0
plane_type=Overlay
ZPOS=1
output_data_type=image:argb8888
]
[DRMMEDIA][I]: open /dev/dri/card0 = 4
[DRMMEDIA][I]: suitable ids: 
	connector:	56
	encoder:	55
	crtc:	53
	plane:	54
[DRMMEDIA][I]: conn id:[56], enc id:[55], crtc id:[53], plane id:[54], w/h:[1920,720], fps:[60]
[DRMMEDIA][I]: OutputStreamFlow:drm_output_stream, thread_model=1, mode_when_full=3
[New LWP 3009]
[DRMMEDIA][I]: Enable vo channel:[1] finished......
[DRMMEDIA][I]: flow-name OutputStreamFlow:drm_output_stream
[DRMMEDIA][I]: Opened DRM device /dev/dri/card0: driver rockchip version 2.0.0
[DRMMEDIA][D]: alloc_drm_memory: ptr:0xa5b24000, fd:6, handle:1, dev_fd:5, size:1228800
[DRMMEDIA][D]: alloc_drm_memory: ptr:0xa59f8000, fd:7, handle:2, dev_fd:5, size:1228800
[New LWP 3010]
[New LWP 3011]
[New LWP 3012]

Thread 4 "azure_gui" received signal SIGUSR1, User defined signal 1.
[Switching to LWP 3010]
0xa6f95bfc in do_futex_wait.constprop () from /lib/libpthread.so.0
(gdb) bt
#0  0xa6f95bfc in do_futex_wait.constprop () from /lib/libpthread.so.0
#1  0xa6f95d38 in __new_sem_wait_slow.constprop.1 () from /lib/libpthread.so.0
#2  0x0007a9e4 in _tx_linux_timer_interrupt ()
#3  0xa6f8c390 in start_thread () from /lib/libpthread.so.0
#4  0xa6ce872c in ?? () from /lib/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
[image]

jiale-gdyd, 2022-10-18:
Hi,
    I am currently using Debian11, using X11 display driver, porting yes threadx is threadx/ports_smp/linux/gnu/, The case used by guix is guix/samples/demo_guix_industrial/. I am debugging using project github address is [https://github.com/jiale-gdyd/gui_project.git]. You can clone the project and run [./build.sh host_azure_gui] in the root directory of the project to generate the azure_gui executable file. Thanks!

jdeere5220, 2022-10-18:
OK that you for the additional information. We are closing out our next patch release but this is first on my list as soon as we wrap up the current update.

ting-ms, 2023-08-28:
Closing the issue. If further issues arise, feel free to reopen it.

closed 2023-08-28 by ting-ms