eclipse-threadx/guix #50: GUIX Studio generates incorrect call signature for GX_STRING_SCROLL_WHEEL_PROPERTIES

RodneyRichardson, 2022-10-31:
Using GUIX Studio 6.2.0 (also tested on 6.1.12):
* create a new project, 
* add a string scroll wheel to the main window
* generate the specification files

GUIX Studio generates a `GX_STRING_SCROLL_WHEEL_PROPERTIES` struct in the `specifications.h` file. The `callback` signature incorrectly contains a `GX_NUMERIC_SCROLL_WHEEL_STRUCT`.

```
typedef struct
{
    int                 total_rows;
    int                 selected_row;
    GX_VALUE            row_height;
    GX_UBYTE            start_alpha;
    GX_UBYTE            end_alpha;
    GX_RESOURCE_ID      normal_font;
    GX_RESOURCE_ID      selected_font;
    GX_RESOURCE_ID      normal_text_color;
    GX_RESOURCE_ID      selected_text_color;
    GX_RESOURCE_ID      disabled_text_color;
    GX_RESOURCE_ID      wallpaper_id;
    GX_RESOURCE_ID      selected_background;
    GX_CONST GX_RESOURCE_ID  *string_id_list;
    UINT (*callback)(struct GX_NUMERIC_SCROLL_WHEEL_STRUCT*, INT, GX_STRING *);
} GX_STRING_SCROLL_WHEEL_PROPERTIES;
```
It should be:
```
    UINT (*callback)(struct GX_STRING_SCROLL_WHEEL_STRUCT*, INT, GX_STRING *);
```

This can be seen in one of the tutorials: https://github.com/azure-rtos/guix/blob/master/tutorials/demo_guix_scroll_wheel/demo_guix_scroll_wheel_specifications.h#L141

jdeere5220, 2022-10-31:
Hello @RodneyRichardson,

Thank you for reporting this! I have duplicated the result in my office. We will get this fixed and get an update pushed asap.

Best Regards,

Ken

jdeere5220, 2022-11-04:
Hello @RodneyRichardson ,

Just to let you know the fix has been merged and an update is being pushed to the App Store. You should received the udpated 6.2.0.1 release automatically within a day or two. I'm going to go ahead and close the ticket, but feel free to re-open it if the 6.2.0.1 patch doesn't address your issue.

closed 2022-11-04 by jdeere5220

RodneyRichardson, 2022-12-06:
@jdeere5220 Hi Ken, Any idea when version 6.2.0.1 will be released on Microsoft Store? or where I can find release notes for the applications?

jdeere5220, 2022-12-06:
@RodneyRichardson the update is on the Store now. Sometimes it takes a while to automatically update, I don't really know how that works, but if you don't see the update within the next 24 hours you can try going back to the store page and see if the "Get" button is enabled to force an update. Mine hasn't auto-updated yet either, but the App Store indicates the updated submission is validated and it should be rolling out.

Regarding release notes, we don't actually publish release notes for GUIX Studio or TraceX today. We keep those internally, but there is no way to get them from the App Store. We are discussing adding those to our github repo so that people have access, though it's not real obvious where to find them. Let me find out if that plan is moving forward and when.

Best Regards,

Ken

RodneyRichardson, 2023-04-11:
Confirmed fixed in 6.2.1.1.