eclipse-threadx/guix #51: _gx_system_font_8bpp Define and declare issues

jiale-gdyd, 2022-11-07:
1.In line common/src/gx_system_font_8bpp.c:1133, GX_FONT _gx_system_font_8bpp = {GX_FONT_FORMAT_8BPP, 0, 0, 18, 13, 0x20, gx_font_format_8bpp, 0, 0, 18, 13, 0x20, 0x7e, {font_SansPro_Regular_ttf_18_aa}, NULL };
2.In line common/src/gx_binres_theme_load.c:34, it is extern GX_CONST GX_FONT _gx_system_font_8bpp;
3.It is also extern GX_FONT _gx_system_font_8bpp in the xxxx_resources.c generated using Guix Studio;
4.Why is it not defined as`GX_CONST GX_FONT _gx_system_font_mono and _gx_system_font_4bpp = {GX_FONT_FORMAT_8BPP, 0, 0, 18,  13, 0x20, 0x7e, {font_SansPro_Regular_ttf_18_aa}, NULL }; `

jiale-gdyd, 2022-11-07:
Why not defined `GX_CONST GX_FONT _gx_system_font_8bpp = { GX_FONT_FORMAT_8BPP, 0, 0, 18,  13, 0x20, 0x7e, {font_SansPro_Regular_ttf_18_aa}, NULL };`

jdeere5220, 2022-11-07:
Certainly looks like it should be declared const in all cases, but let me look into it a little further and get back to you. Thanks for the input.

ting-ms, 2023-07-04:
The fix has been pushed to GitHub.

closed 2023-07-04 by ting-ms