eclipse-threadx/guix #52: Can GUIX Studio port boldtype font ?

flyflyfly001, 2022-11-28:
Hi, I have a .ttf  font package which contains several font type like bold/medium/regular/light and so on. Can I use GUIX Studio to port a boldtype font?  Or generate a runtime bold style from regular style?

jdeere5220, 2022-11-28:
Can you send me the font please? Usually when a font contains several styles, it has the extension .ttc or "True Type Collection". In that case I believe GUIX Studio will allow you to choose a specific typeface from the collection. There isn't any ability to modify the font at runtime, so you need to produce a bold font using GUIX Studio.

flyflyfly001, 2022-12-03:
I'm using GUIX Studio 6.2.0.0. The font with .ttc extension is being filtered when I click _Add New Font_ in the file browser.

closed 2022-12-10 by flyflyfly001