eclipse-threadx/guix #53: GUIX Studio does not generate proper speficiation code when JPEG support is disabled

int-bio, 2022-12-14:
**Describe the bug**
We are unable to compile gui_specifications.c without having to modify the file after we have regenerated the file.  

The issue surrounds the setting up of gx_display_driver_jpeg_draw.  We disabled JPEG support in e2 studio.  gx_display_driver_jpeg_draw is compiled out correctly in gx_display_driver_synergy_dave2d.c, however the flag is being ignored in gui_specifications.

Please also mention any information which could help others to understand
the problem you're facing:
- Building for cortex_m4
- Azure tx_api 6.1.10
- What toolchain and environment?
      - e2 Studio 2022-04 (22.4.0)
      - guix studio 5.4.2.9
- What have you tried to diagnose or workaround this issue?
      - We have developed a script to replace code (see additional context below) in the file after we have generated new output files from GUIX Studio.

**To Reproduce**
Steps to reproduce the behavior:
1. Open e2 studio configuration file.
2. Add GUIX stack
3. Add GUIX source
4. Disable Synergy JPEG support
5. Rebuild project 

**Expected behavior**
The code should compile without warnings or errors.  We assume the pointer will be NULL.

**Impact**
As of today, we can get by with a script that patches the file. (or hand edit).

**Logs and console output**
```
../src/gui/gui_specifications.c: In function '_gx_synergy_display_driver_setup':
'gx_display_driver_jpeg_draw'; did you mean 'gx_display_driver_polygon_draw'?
18413 |     display -> gx_display_driver_jpeg_draw                     = _gx_synergy_jpeg_draw;
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
```

**Additional context**
You should add in PNG flags in case someone decides that they do not require PNG support. #define GX_SOFTWARE_DECODER_SUPPORT

The following code will fix the issue:
```
#if (GX_USE_SYNERGY_JPEG == 1)
    display -> gx_display_driver_jpeg_draw                     = _gx_synergy_jpeg_draw;
#endif
#if defined(GX_SOFTWARE_DECODER_SUPPORT)
    display -> gx_display_driver_png_draw                      = _gx_display_driver_565rgb_png_draw;
#endif
```

int-bio, 2022-12-14:
[image]

jdeere5220, 2022-12-15:
@int-bio can you tell me your GUIX Studio project settings as well? In the project configuration dialog, when you select Synergy as the target CPU, there is an advanced settings page that allows you to enable/disable the Dave2D accelerator as well as the hardware jpeg decoder. Let me know your project settings here as well please.

int-bio, 2022-12-19:
[image]

Here are the first few lines up to languages.
```<project_version>55</project_version>
<guix_version>50401</guix_version>
<studio_version>5040209</studio_version>
<project_name>gui</project_name>
<source_path>.\</source_path>
<header_path>.\</header_path>
<resource_path>.\</resource_path>
<insert_headers_before>FALSE</insert_headers_before>
<target_cpu>1</target_cpu>
<target_tools>2</target_tools>
<big_endian>FALSE</big_endian>
<synergy_graph_accelerator>TRUE</synergy_graph_accelerator>
<synergy_jpeg_decoder>2</synergy_jpeg_decoder>
<synergy_png_decoder>1</synergy_png_decoder>
<grid_enabled>FALSE</grid_enabled>
<snap_enabled>FALSE</snap_enabled>
<grid_spacing>10</grid_spacing>
<snap_spacing>10</snap_spacing>
<gen_binary>FALSE</gen_binary>
<binary_file_format>1</binary_file_format>
<memory_offset>0</memory_offset>
<gen_res_header>TRUE</gen_res_header>
<custom_resource_enabled>FALSE</custom_resource_enabled>
<app_execute_xpos>0</app_execute_xpos>
<app_execute_ypos>0</app_execute_ypos>
<is_widget_position_locked>FALSE</is_widget_position_locked>
<num_displays>1</num_displays>
<max_displays>4</max_displays>```

I noticed that <synergy_jpeg_decoder>2</synergy_jpeg_decoder> is set, but there is no widget to change that.

jdeere5220, 2022-12-19:
@int-bio you need to click on the Advanced Settings button and turn off the hardware jpeg decoder: 
[image]

The settings in your Eclipse/e2Studio project need to match the settings in your GUIX Studio project, and unfortunately they are not automatically synched. So if you are turning off the hardware jpeg decode ip, then you also need to turn this off in the GUIX  Studio project. Let me know if that takes care of the issue.

int-bio, 2022-12-19:
Not a bug, operator error.  Thank you.

closed 2022-12-19 by int-bio