eclipse-threadx/guix #56: Unable to import into latest GUIX Studio

int-bio, 2022-12-19:
**Describe the bug**
We are able to open and export a GXP project in GUIX Studio version 5.4.2.9.  However, when we attempt to open the same project file in Azure RTOS GUIX Studio version 6.2.0.1, the application aborts (it is killed) without any notifications.  

Please also mention any information which could help others to understand
the problem you're facing:
   -  Building for cortex_m4
    - Azure tx_api 6.1.10
    What toolchain and environment?
    - Azure RTOS GUIX Studio 6.2.0.1
    What have you tried to diagnose or workaround this issue?
    - tried using the -o output.log

**To Reproduce**
Steps to reproduce the behavior:
1. Open GXP file using version 6.2.0.1
2. See error

**Expected behavior**
Project file is read and a project screen is displayed

**Impact**
We are unable to use the product. 

**Logs and console output**
The -o option does not provide any clues.

**Additional context**
If you can provide me with a method to give you better verbose, I will try. it.

jdeere5220, 2022-12-19:
@int-bio well obviously that should never happen. This one sounds like a bug. We have added a bunch of error checking to prevent malicious projects from allowing bad actors to access your machine, which is part of the continuous and ongoing Microsoft security focus, but GUIX Studio should tell you WHY it doesn't like your project. Any chance you can send me the original 5.4.3.9 project? We do lots of regression testing, so I don't think this is a general bug, but something specific to the content of your project. It's going to be difficult for me to nail it down unless you can send your .gxp file to me.

int-bio, 2022-12-19:
I would be happy to send you the gxp file.  Is there some place other then the public forum to send it to?

jdeere5220, 2022-12-19:
@int-bio please attach it to an email to azure-rtos-support@microsoft.com. Thanks!

jdeere5220, 2022-12-20:
@int-bio I was able to duplicate the error you are seeing. It looks like your project is exploiting a bug in the earlier version of GUIX Studio, i.e. you have lots of different widgets defined with the same widget name, like "win_calibrate" or "win_timers". That's actually not allowed, but there was a bug in the older version of GUIX Studio that allowed it. So at this point I'm wondering about the best way to fix it. We could force GUIX Studio to make your widget names unique when the project is loaded by appending something to each duplicate name to make them unique, but that's going to affect the generated specifications file and it could possible break your application code if you are referencing these widget names directly, which you likely are doing. I really do not want to break your application code! I'm open to suggestions.

int-bio, 2022-12-20:
IMHO: The program should just output it's issues and make me fix it.  
     ie: duplicate widget name "win_calibrate".

jdeere5220, 2022-12-20:
Sure, I was just looking into it a bit further. We definitely have to fix the issue, it's no good having Studio just go off in the weeds because it gets confused about widget names. I will get back to you once we implement a fix. With the holidays it's going to take a few days.

int-bio, 2022-12-20:
Understood.  Thank you

int-bio, 2022-12-21:
Should I be looking at <app_name> for duplicates?   I am not sure why win_timers is an issue. I only see it defined in one location and referenced as a template in another.

int-bio, 2022-12-21:
Good news, I was able to update our GXP file to get past the initial startup. However, when I select the patient screen, the application exits.  I will email you my updated file.

int-bio, 2022-12-22:
I was able to knock out the last two mis-formed widget names and I am able to use the tool now.

ting-ms, 2023-07-04:
GUIX Studio version 6.2.1.3 has been released on the Microsoft Store, and it includes the fix.

closed 2023-07-04 by ting-ms