eclipse-threadx/guix #57: gx_prompt_text_id_set doesn't conform to transparency setting

int-bio, 2022-12-29:
**Describe the bug**
We are porting from SSP 1.5.2.0 -> 2.3.0.0, using GUIX 6.2.0.1

On the initial splash screen, our prompt widget shows up correctly.  WHITE text, transparent on a dark wallpaper background.
<see good.png>

However, when we issue a gx_prompt_text_id_set() , the prompt widget shows WHITE text on a WHITE background.  Initially, I thought that the transparency seems to have been ignored, but if it had not been used, the background would be BLACK and not WHITE.
<see bad.png>

Here is the prompt widget configuration prior to calling gx_prompt_text_id_set:

```wid    GX_PROMPT *    0x2005ce68 <splash_screen_3+232>    
    gx_widget_name    const GX_CHAR *    0x1781ec "selftest_str"    
    gx_widget_parent    struct GX_WIDGET_STRUCT *    0x2005cd80 <splash_screen_3>    
    gx_widget_first_child    struct GX_WIDGET_STRUCT *    0x0 <R_SSP_VersionGet>    
    gx_widget_last_child    struct GX_WIDGET_STRUCT *    0x0 <R_SSP_VersionGet>    
    gx_widget_next    struct GX_WIDGET_STRUCT *    0x2005cee0 <splash_screen_3+352>    
    gx_widget_previous    struct GX_WIDGET_STRUCT *    0x2005cdf4 <splash_screen_3+116>    
    gx_widget_nav_next    struct GX_WIDGET_STRUCT *    0x0 <R_SSP_VersionGet>    
    gx_widget_nav_previous    struct GX_WIDGET_STRUCT *    0x0 <R_SSP_VersionGet>    
    gx_widget_size    GX_RECTANGLE    {...}    
    gx_widget_clip    GX_RECTANGLE    {...}    
    gx_widget_event_process_function    UINT (*)(struct GX_WIDGET_STRUCT *, GX_EVENT *)    0x3cd1d <_gx_prompt_event_process>    
    gx_widget_draw_function    void (*)(struct GX_WIDGET_STRUCT *)    0x3ccf9 <_gx_prompt_draw>    
    gx_widget_status    ULONG    0x10000003 (Hex)    
    gx_widget_style    ULONG    0x50004000 (Hex)    
    gx_widget_normal_fill_color    GX_RESOURCE_ID    0    
    gx_widget_selected_fill_color    GX_RESOURCE_ID    0    
    gx_widget_disabled_fill_color    GX_RESOURCE_ID    0    
    gx_widget_type    USHORT    30    
    gx_widget_id    USHORT    216    
    gx_widget_user_data    INT    0    
    gx_widget_alpha    GX_UBYTE    0xff (Hex)    
    gx_prompt_string    GX_STRING    {...}    
    gx_prompt_text_get_function    void (*)(struct GX_PROMPT_STRUCT *, GX_STRING *)    0x3ce71 <_gx_prompt_text_get_ext>    
    gx_prompt_text_id    GX_RESOURCE_ID    120    
    gx_prompt_font_id    GX_RESOURCE_ID    6    
    gx_prompt_normal_text_color    GX_RESOURCE_ID    6    
    gx_prompt_selected_text_color    GX_RESOURCE_ID    37    
    gx_prompt_disabled_text_color    GX_RESOURCE_ID    6   ```

After the call, the only part of the structure that has changed is the gx_prompt_text_id().  Both gx_widget_status  and gx_widget_style    stay the same (indicating transparency is enabled).

Please also mention any information which could help others to understand
the problem you're facing:
- Building for cortex_m4
- Azure tx_api 6.1.10
- Azure gx_api 6.1.10
- Azure RTOS GUIX Studio 6.2.0.1

** What have you tried to diagnose or workaround this issue?
- gx_prompt_text() does the same thing.

**To Reproduce**
Steps to reproduce the behavior:
1. I have provided the setup of the text widget. Not sure what else I can give you here.

**Expected behavior**
Text should be WHITE on a wallpaper background

**Impact**
Showstopper, can't deliver code.

**Logs and console output**
I have attached the before and after displays.

**Additional context**
Add any other context about the problem here.
![bad_splash png-640x480](https://user-images.githubusercontent.com/108280402/209960501-6a10bd66-75ee-45b3-8503-ff5b59a7420c.png)
![good_splash png-640x480](https://user-images.githubusercontent.com/108280402/209960503-00406947-08c3-4c61-94ff-afcd189a4ffd.png)

jdeere5220, 2022-12-29:
@int-bio I wrote a little test using our demo_guix_widget_types and it seems that it is working as it should. A little bit of detail:
1) demo_guix_widget_types inclues a screen with a bunch of different text widgets, including transparent prompts and a numeric prompt.
2) I added a custom drawing function to Text_Screen to draw a pixelmap wallpaper background.
3) The numeric prompt still works, it's updating and the pixelmap shows through.
4) I modified "Transparent prompt without border" so that when you click on it I assign a new text id using gx_prompt_text_id_set. That also works as expected: 
[image]

I changed the text to "September" using the gx_prompt_text_id_set() api.

As to HOW it works, if you look at the function gx_system_canvas_refresh(), in the file of the same name, around line 145 we check to see if the widget to be re-drawn is transparent. If it is, we back up to the nearest non-transparent widget and invoke that widget's drawing function (with the same invalid rectangle). This is how the prompt background gets erased, if you will, when you change the text. Then the prompt draws the new text over this blank background.

So I think it should be working, and we might need to step into your application code and see what is different than my test case. So far I don't see anything wrong. The best case would be if you can send me a test case that I can build that duplicates the problem. If you like I can send you my modified demo_guix_widget_types so that you can confirm a transparent prompt should indeed be working.

jdeere5220, 2022-12-29:
@int-bio one more thing we could check, you could step into the gx_prompt_draw() function. It will eventually end up at a generic gx_widget_border_draw(), which is where the widget border and background are drawn. At line 231 of gx_widget_border_draw.c, we test to see if the widget is transparent. If so, we don't invoke the gx_canvas_rectangle_draw() API, i.e. we don't fill the widget so that background should show through. I'm curious what happens when you step through this code.

int-bio, 2022-12-30:
We have a simple setup here.  The following window is created under root.

-> (win)
[image]

Prompt is created under (win)
-> (prompt) 

[image]

I am trying to set a break point to follow changes when the widget-id = ID_SELFTEST_STR

int-bio, 2022-12-30:
After thinking about it issue, this may be an artifact of a different issue we are having.  In our current boot sequence, we go from splash-screen-1 to splash-screen-2 (self-test) to main screen.  During the transition from splash-screen-2 to main, our wallpaper transitions from the bitmap (dark) to all white. 

I had thought this wallpaper issue was due to an invalid widget in the main screen, but what if the issue is actually occurring in splash-screen-2 and only shows up when we issue the text update.  The wall paper image gets cleared out (or the pointer to wallpaper points to a different location) and so the update pulls the bitmap region from the wrong location and stuffs it with WHITE.  I will be checking into this a little deeper.

int-bio, 2023-01-04:
I was able to trace the bug back to one of our modules.