eclipse-threadx/guix #59: Documentation question about gx_system_event_send() error code

int-bio, 2023-02-13:
gx_system_event_send(GX_EVENT * event) returns GX_SYSTEM_ERROR (0xFE) as a return error code.  
The online [documentation](https://learn.microsoft.com/en-us/azure/rtos/guix/chapter-4/) does not state that this is a valid error code.  Can you explain what this error code may mean?

I have verified that:
```
- GX_EVENT * is not NULL
- event:
    gx_event_type	ULONG	0x40000020 (Hex)	
    gx_event_display_handle	ULONG	0	
    gx_event_target	struct GX_WIDGET_STRUCT *	0x0 <R_SSP_VersionGet>	
    gx_event_sender	USHORT	0	
    gx_event_payload	union {...}	{...}	
	gx_event_ulongdata	ULONG	0	
```

jdeere5220, 2023-02-13:
Hello @int-bio,

I apologize that the documentation does not indicate that this is a possible return code. I have created a backlog task to get that fixed.

Regarding this error return, I need to know if you are running with ThreadX or another RTOS binding. Evaluating the source code for gx_system_event_send(), the only path that can return an error code is if the call to tx_queue_send() fails (if running with ThreadX) or the GX_EVENT_PUSH macro/function function returns an error (if running with another RTOS).

So we really need to look at the possible reasons why tx_queue_send might fail (with no wait timeeout). The most likely cause is the event queue is full (i.e. you are pushing events faster than GUIX can process them). But there are other possible causes, such as calling this function from an ISR. I would refer you to the ThreadX documentation of the tx_queue_send function.

Best Regards

int-bio, 2023-02-13:
GUIX Studio 6.2.0.2
SSP 2.3.0
GUIX Library 6.1.10

int-bio, 2023-02-13:
I do believe it is possible that the issue is a full queue based on another issue I have place with Renesas. 

https://community.renesas.com/mcu-mpu/embedded-system-platform/f/forum/29854/display-glitches-constantly-when-d-ave-2d-frame-buffer-cache-is-enabled

jdeere5220, 2023-02-13:
@int-bio to confirm, break in the debugger when you receive the error code, and examine the data structure _gx_system_event_queue. Send me a screen shot if you like. We should be able to confirm based on the fields in this control structure.

jdeere5220, 2023-02-13:
Yep, the queue is full. Have to throttle back the event sender or give GUIX higher priority.

ting-ms, 2023-07-04:
The online documentation has been updated to address the mentioned issue.

closed 2023-07-04 by ting-ms