eclipse-threadx/guix #61: Bad LowerCase/UpperCase file loading in recent file window

xNiux, 2023-05-26:
**Describe the bug**
When you loading a project with Recent Project window or menu, the project file is loading in lowercase instead of real case name.
When you generate files, the project name is used with the loaded case name instead real case name

Please also mention any information which could help others to understand
the problem you're facing:
- GuiX Studio version : 6.2.1.2
- Environment : Windows 10

**To Reproduce**
Steps to reproduce the behavior:
1. Create a project with different lower/upper case like "GuixProject.gxp"
2. Open It  with a double click on the file in file explorer : File is correctly loaded (GuixProject.gxp)
3. Generate files : All reference of project name are "GuixProject_****"
4. Close Guix Studio
5. Open Guix Studio, the Recent Projects window will list all project .... In lower case !
6. Open with Recent Projects Window your "GuiProject.gxp" : File is lowercase loaded (guixproject.gxp)
7. Generate files : All reference of project name are "guixproject_****"

**Impact**
Compilation with error because of bad variable/constant name (C is case sensitive)

[image]
[image]
[image]
[image]
[image]
[image]

ting-ms, 2023-05-29:
Hi xNiux, I'm able to reproduce the issue, and we'll fix it.

ting-ms, 2023-05-31:
Hi xNiux, we have fixed the issue, and will be included in the next release.

zhutnana, 2023-07-04:
GUIX Studio version 6.2.1.3 has been released on the Microsoft Store, and it includes the fix.

ting-ms, 2023-07-04:
GUIX Studio version 6.2.1.3 has been released to Microsoft Store, and it includes the fix.

closed 2023-07-04 by ting-ms