eclipse-threadx/guix #62: GUIX Studio RGB888 frame buffer support

flyflyfly001, 2023-06-05:
Hi there,

I'm using a 640*480 resolution lcd display with RGB888 color format, so the frame buffer size shoud be 640*480*3 = 921600 byte.
But the GUIX Studio generated canvas memory size is 1228800 byte, which is 640*480*4. It means each pixel contains 4 byte.
(GUIX Studio Version: 6.2.1.2)
[image]
[image]

Question 1: 
Why each pixel contains 4 byte when I choose 24bpp ?

Question 2: 
Because of hardware memory limitation, the canvas size shound not exceed 1M byte. So each pixel must be  RGB888 format without alpha channel. Can GUIX support this ?

Any help would bu much appreciated.

ting-ms, 2023-06-05:
The 24bpp here represents 24xrgb with the first 8 bits being ignored, and the remaining 24 bits used for the RGB (Red, Green, Blue) color channels.

The RGB888 format is not currently supported.

flyflyfly001, 2023-06-06:
> The 24bpp here represents 24xrgb with the first 8 bits being ignored, and the remaining 24 bits used for the RGB (Red, Green, Blue) color channels.
> 
> The RGB888 format is not currently supported.

But I see GX_COLOR_FORMAT_24RGB packed mode here. Does GUIX have any plan to support this ?
[image]

ting-ms, 2023-06-06:
We currently do not have a plan to support this format. I would like to offer two recommendations: 
1. Use 565rgb format.
2. Implement your own 24rgb format display drivers, this might require some effort on your part. If you choose to pursue this route, I am willing to provide guidance.

flyflyfly001, 2023-06-07:
> 

Hi ting,

My lcd display dose not support RGB565 format. So I need to implement my own 24rgb format. Could you give some brief illustration about where and how many code I need to midify ? 
Thanks.

ting-ms, 2023-06-08:
Here's a list of format-related driver functions that need to be implemented:
**_gx_display_driver_24rgb_setup:** Function for setting display drivers. Drivers not supported can be set to GX_NULL. You can refer to the definitions of other driver setup functions, such as **_gx_display_driver_24xrgb_setup**. 
**_gx_display_driver_24rgb_horizontal_line_draw:** Draws a horizontal line.
**_gx_display_driver_24rgb_vertical_line_draw:** Draws a vertical line.
**_gx_display_driver_24rgb_pixel_write:** Writes a pixel to the canvas with the specified coordinates.
**_gx_display_driver_24rgb_block_move:** Shifts a block of canvas memory
**_gx_display_driver_24rgb_row_pitch_get:** Calculates the row pitch in bytes for a canvas the of given width.
**_gx_display_driver_24rgb_pixelmap_draw:** Draws a pixelmap in the 24rgb pixelmap. Since GUIX Studio does not support this format yet, you can define your own pixelmap format and implement the drawing accordingly. 
**_gx_display_driver_24rgb_pixel_blend:** Wirtes a pixel to the canvas with alpha blending.

The following functions may not be needed depending on your specific requirements:
**_gx_display_driver_24rgb_1bit_draw:** Not needed if the 1-bit font is not used.
**_gx_display_driver_24rgb_canvas_copy:** Not needed if only one canvas is used.
**_gx_display_driver_24rgb_canvas_blend:** Not needed if there is no need for canvas blending.
**_gx_display_driver_24rgb_pixelmap_blend:** Not needed if there is no need for pixelmap blending.

May I ask which hardware you are using?
Are you in the stage of evaluation or have you already get started production?
At this time, we will carefully consider the feasibility and potential benefits of incorporating this format into our system.
While we do not have immediate plans for its implementation, we will keep it under consideration for future updates and enhancements.

flyflyfly001, 2023-06-09:
Hi ting,

We are using Apollo4 SOC for evaluation now.

ting-ms, 2023-06-15:
Thank you for your valuable input, and we will continue to evaluate the possibilities associated with this format going forward.

closed 2023-06-15 by ting-ms