eclipse-threadx/guix #65: GUIX in low power mode

flyflyfly001, 2023-09-15:
As we know, GUIX event is driven by a timer. Which is 30Hz or higher for refreshing UI. But when the lcd is power off. We do not need to refresh it，the CPU is also in sleep mode. Can I directly change the timer frequency to avoid CPU wakeup ? or any other better method for saving power consumption?

ting-ms, 2023-09-15:
@flyflyfly001 What kind of GUIX event do you mean? By default, if the application hasn't created any timers, the system timer will remain inactive. Could you please provide more details about your particular use case for clarification?

flyflyfly001, 2023-09-19:
Yes, the application has already created many timers. But when the lcd is power off, we cannot stop all the timers created before. The reason is first, we do not have an API to turn off all the timers the same time. Second, timer start and stop logic is correspond to application. Cannot simply turn off it when lcd is power off.

ting-ms, 2023-09-20:
@flyflyfly001 You are able to call `gx_system_timer_stop(widget, timer_id)` to stop a timer associated with a widget, using the specified timer ID. Alternatively, you can stop all the timers belonging to a widget with the timer ID set to 0. 

If you do not want to stop all the timers when the LCD is powered off, I suggest you add application logic to prevent sending events to the GUIX system when a timer event is received. This will help in avoiding unnecessary canvas refreshing.

flyflyfly001, 2023-10-17:
if we set GX_TICKS_SECOND to 50, then guix timer task will be running at 50Hz. So every 20ms CPU will be waked up to check if  any events need to be processed. If there is no events, CPU still need to wakeup to check. Am I right ?

ting-ms, 2023-10-18:
The system timer is not used for checking input events, it is used for trigger timer events for timers and updating animations if any are in progress. Input events, on the other hand, are managed by the GUIX system thread. If no events are present, the GUIX system thread enters a blocked state.

flyflyfly001, 2023-10-18:
Yes, _gx_system_thread_entry_shell_ will enter block state. But the guix timer task _guix_timer_task_entry_ still works at 50Hz， it will wake up the CPU at 50Hz. I have doing some tests and it can be confirmed.

ting-ms, 2023-10-18:
Do you have application-defined timers running? If that is the case, yes the timer thread would still work.

ting-ms, 2023-10-19:
The system timer thread is suspended upon creation, and it becomes active once you call _gx_system_timer_start_ to start a timer for a specific widget. If all timers are stopped, the system timer thread will go to the suspended state again.

flyflyfly001, 2023-10-20:
Ok, I got it. Thanks

ting-ms, 2023-10-27:
Closing the issue.

closed 2023-10-27 by ting-ms