eclipse-threadx/guix #67: GUIX Studio font heights above 240 pixels are not possible

IvanThor, 2023-11-15:
Setting the font heights in GUIX studio to more than 240 pixels, e.g. 255 pixels, results in faulty behaviour. The font view doesn't show the font anymore. Moreover the font isn't shown when used in a prompt widget.
If the generated project is used with the actual hardware, the faulty behaviour stays. The font is not shown.

GUIX Studio 6.3.0.0

Steps to reproduce the behavior:
1. Create a new GUIX studio project
2. Add a custom font to the project and set font heigth to 255 pixels.

ting-ms, 2023-11-16:
@IvanThor GUIX keeps track of a metric within the GX_FONT structure called gx_font_line_height. This metric represents the sum of the maximum ascent and the maximum descent of the font glyphs. It's possible that the line height can exceed the font height, as explained in more detail at https://learn.microsoft.com/en-us/azure/rtos/guix/guix-studio-4#font-resources.  In your case, when the font height is set to a value greater than 240, the resulting line height may surpass the 255 limit of the line height. We'll add a note about this in the future.

IvanThor, 2023-11-16:
@ting-ms Disabling ASCII as character range and just include character set defined by String Table fixed the issue indeed.

[image]

Thank you very much

closed 2023-11-16 by IvanThor