eclipse-threadx/guix #81: If a vertical lists has a multiple childs of different heights, the scrollbar thumb does not reach the end of the scrollbar

franlovera1, 2023-11-29:
**Describe the bug**
If a vertical lists has a multiple childs of different heights, the scrollbar thumb does not reach the end of the scrollbar.
So for example if a vertical lists has these elements:
[image]
 the scrollbar thumb does not reach the bottom of the scrollbar.
_gx_vertical_list_scroll_info_get() only calculates the scroll maximum correctly if all the childs are the size of list->gx_vertical_list_child_height. If the childs has varying heights, the calculation is incorrect.
A suggested fix would be to iterate though all the childs and add the height of each child one by one:
[image]

Please also mention any information which could help others to understand
the problem you're facing:
- What target device are you using? STM MCU
- Which version of Azure RTOS? 6.3.0
- What toolchain and environment? clang
- What have you tried to diagnose or workaround this issue? yes see proposed solution

**To Reproduce**
Steps to reproduce the behavior:
1. create a vertical list with childs of different heights.
2. Add a scroll bar to the vertical lists, set size to relative,
3. Run code. The scrollbar does not reach the bottom

**Expected behavior**
When the screen is fully scrolled, the scrollbar thumb does not reach the bottom of the bar.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper): annoyance

**Logs and console output**
If applicable, add console logs or other types of debug information like Wireshark capture as `.zip` file.

**Additional context**
Add any other context about the problem here.

ting-ms, 2023-12-01:
@franlovera1 Thank you for your feedback, we'll investigate the issue.

ting-ms, 2023-12-19:
@franlovera1 This issue has been fixed in this PR: https://github.com/azure-rtos/guix/pull/93

closed 2023-12-19 by ting-ms

franlovera1, 2023-12-19:
Thanks, I'll take a look!