eclipse-threadx/guix #82: The GX_EVENT_FOCUS_LOST event is not sent if the parent does not have focus

maxim-lbmv, 2023-12-03:
**Describe the bug**
The GX_EVENT_FOCUS_LOST event is not sent if the parent does not have focus

For example, there is the following hierarchy

- Widget 1
  - Widget 1.1
    - Widget 1.1.1
  - Widget 1.2

Widget 1.1 does not accept focus. Focus on Widget 1.1.1. If you click on Widget 1.2, then Widget 1.1.1 will not receive the GX_EVENT_FOCUS_LOST event. This happens because when focus is transferred, there is a widget with a common parent, in our case it is Widget 1.1, and the focus loss event is passed to it, but it does not pass it on because it does not accept focus.

Please also mention any information which could help others to understand
the problem you're facing:
- What target device are you using? Cortex M7
- Which version of Azure RTOS? 6.1
- What toolchain and environment? IAR
- What have you tried to diagnose or workaround this issue?  We have to make sure that all widgets accept focus, but this does not work with vertical lists, because widgets there do not accept focus and we have to write special handlers for them so that they pass the focus loss event to children

**To Reproduce**
1. Create a project with 4 widgets, with hierarchy and settings, as shown in the description
2. Add event processing for GX_EVENT_FOCUS_LOST Widget 1.1.1
3. Click on Widget 1.1.1
4. Click on Widget 1.2

**Expected behavior**
Widget 1.1.1 will receive the GX_EVENT_FOCUS_LOST event

**Impact**
Widget 1.1.1 does not receive GX_EVENT_FOCUS_LOST event

**Logs and console output**
No

**Additional context**
No

ting-ms, 2023-12-07:
@maxim-lbmv Thank you for your feedback! We can reproduce the issue and we'll fix it in the future.

ting-ms, 2023-12-18:
@maxim-lbmv This issue has been fixed in this PR: https://github.com/azure-rtos/guix/pull/92.

closed 2023-12-18 by ting-ms