eclipse-threadx/guix #83: GX_VERTICAL_LIST does not release input if it contains GX_SINGLE_LINE_TEXT_INPUT

maxim-lbmv, 2023-12-03:
**Describe the bug**
GX_VERTICAL_LIST does not release input if it contains GX_SINGLE_LINE_TEXT_INPUT

This happens because after clicking on GX_SINGLE_LINE_TEXT_INPUT it takes the input and receives the GX_EVENT_PEN_UP event, which should have taken the input from GX_SINGLE_LINE_TEXT_INPUT. As a result, clicks are processed by GX_VERTICAL_LIST and other elements cannot receive focus.

Please also mention any information which could help others to understand
the problem you're facing:

- What target device are you using? Cortex M7
- Which version of Azure RTOS? 6.1
- What toolchain and environment? IAR
- What have you tried to diagnose or workaround this issue? We have to add parents for GX_SINGLE_LINE_TEXT_INPUT located in GX_VERTICAL_LIST that block the GX_EVENT_PEN_DOWN event from passing to GX_VERTICAL_LIST so that it does not capture the input.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a project with GX_VERTICAL_LIST containing GX_SINGLE_LINE_TEXT_INPUT and another widget
1. Click on GX_SINGLE_LINE_TEXT_INPUT
1. Click on another widget

**Expected behavior**
The click event will be handled by another widget

**Impact**
Click events are handled by GX_VERTICAL_LIST

**Logs and console output**
No

**Additional context**
No

ting-ms, 2023-12-07:
@maxim-lbmv Thank you for your feedback! We can reproduce the issue and we'll fix it in the future.

ting-ms, 2023-12-18:
@maxim-lbmv This issue has been fixed in this PR: https://github.com/azure-rtos/guix/pull/91

closed 2023-12-18 by ting-ms