eclipse-threadx/guix #85: Increase maximum font size

IvanThor, 2023-12-05:
Currently I'm working on a project which requires a visible font heigth of at least 40 milli meters. The display I'm using has a resolution of 320x480 pixels, while 1 pixel is 0.153mm in height. With a maximum font size of 255 pixels the maximum possible font height would be 39.015 mm. So I'm not able to meet the projects requirements.

A way to set font size above 255 pixels would be great. Including a way in GuiX studio to increase the size. Currently it is not even allowed to set a size bigger than 255 pixels.

I already removed glyphs by defining characters in a string which are really used in the project. That just gives me 30mm, tops.
Converting the needed characters into images, e.g. png, with a size of 40mm is not an option, because then I won't be able to set the font color.

ting-ms, 2023-12-06:
@IvanThor Recommend converting the needed characters into PNG images. In the GUIX Studio pixelmap edit dialog, set the image as the alphamap format, retaining only alpha information. Assign an alphamap to a widget, and set the widget as transparent, the widget fill color will work as the alphamap drawing color.

IvanThor, 2023-12-07:
Alright, it has to suffice. Thank you very much.

closed 2023-12-07 by IvanThor