eclipse-threadx/guix #89: Hebrew : multiline text view widgets dont display correctly on several lines

ThibaudBrandt, 2023-12-11:
**Describe the bug**
In Hebrew, multiline text display widgets do not display strings correctly on several lines
The display is correct if the string fit on one line, but not correct on several lines
define GX_DYNAMIC_BIDI_TEXT_SUPPORT and use of _gx_system_bidi_text_enable doesnt seem to affect the draw
The problem seems to come from the _gx_multi_line_text_view_text_draw function.
The cutting of the character chain in several lines does not seem to be done by taking the chain in the right order. 
This one should start in the opposite direction in memory
Imagine the string "GFEDCBA" as Hebrew characters. On one line it is display as "GFEDCBA" (store like this in memory)
If the maximum characters per line is 4 :
On two lines it is displayed as (because the text is drawn from left to right)
"GFED
  CBA"
but should be displayed as (from right to left)
"DCBA
 GFE"

Please also mention any information which could help others to understand
the problem you're facing:
- What target device are you using? : RA6M3 from renesas
- Which version of Azure RTOS? 6.2.1, Guix Studio 6.3.0.1
- What toolchain and environment? GCC, Ide e2studio

**To Reproduce**
Steps to reproduce the behavior:
1. Create a page with a multiline text view widget
2. Assign a Hebrew string on this widget. Long enough to be displayed on several lines

**Expected behavior**
The string should be displayed correctly.

**Impact**
multiline text view widget with Hebrew

**Logs and console output**
None

**Additional context**
None

ting-ms, 2023-12-12:
@ThibaudBrandt Have you defined a user-define memory allocator and de-allocator using `gx_system_memory_allocator_set` API? Runtime allocation of memory is required for dynamic bidirectional text drawing.

ThibaudBrandt, 2023-12-12:
@ting-ms : Yes I did. We don't have any allocation issue, except the need to increase it when gx_system_bidi_text_enable was called. The text is displayed correctly if there is only one line, but on severals lines the text is not displayed in the correct order. I also tried to use the "gx_system_bidi_text_enable" function, but I dont see any changes except a slowdown of the display and some string that dont appear anymore.
We managed to display it correctly by reversing the 'string' in the '_gx_multi_line_text_view_text_draw' and '_gx_multi_line_text_view_string_total_rows_compute' functions to force the calculation of the string split across multiple lines, displayed from right to left. However, this doesn't seem optimal, and I believe we might have overlooked something.
I can share the modified files if needed

ting-ms, 2023-12-12:
@ThibaudBrandt Could you please share the Hebrew string along with a screenshot of the displayed results? I would like to test it on my end.

ThibaudBrandt, 2023-12-12:
הכינו את הקרם יום קודם הפרידו את הבצק לריבועים של 5 ס''מ לפני הבישול. הכינו את רוטב הסירופ. פזרו אבקת סוכר. הגישו קר.

@ting-ms : Here is above the string. 

Here is a screen shot with above the string displayed in GUIX studio and bellow the string in the embedded product.
[image]

ting-ms, 2023-12-13:
@ThibaudBrandt The result displayed in GUIX Studio should match those in your embedded product. Would you check the language setting in your GUIX Studio project:
[image]
If you are using runtime bidi text drawing, ensure that the option **Generate Bidi Text in Display Order** is unchecked.

ThibaudBrandt, 2023-12-13:
@ting-ms : I tried to unchecked the **Generate Bidi Text in Display Order**.  It works. However, the real-time rendering is considerably slowed down, and the strings sporadically disappear. I think checking the option 'Generate Bidi Text in Display Order' and correctly cutting and displaying strings is the correct way to do it in our case

ting-ms, 2023-12-14:
@ThibaudBrandt The checkbox **"Generate Bidi Text in Display Order"** instructs GUIX Studio to generate bidirectional text to the output file in its display order. If this option is selected, no runtime processing is required within the GUIX library to properly render bidirectional text. When this option is selected, bidirectional text rendering should NOT be enabled within the GUIX library. This configuration yields the best runtime performance, but does not support rendering of dynamically defined bidirectional text strings. 

In general, the bidi text reordering process is paragraph based without line breaking. To render a bidi text in a multiline text view widget with appropriate line breaks, assign the string to the designated multi-line text view widget within GUIX Studio. Subsequently, GUIX Studio will generate a reordered bidi text for the multi-line text view, incorporating the necessary line breaking.

ThibaudBrandt, 2023-12-14:
@ting-ms: All our strings are defined in GUIX Studio with an ID but not necessarily assigned to a widget in particular. I tried to assign the string to the widget it's supposed to be displayed on. However, it is not very convenient because I had to invert the language index to assign the number 1 to Hebrew in order to assign the Hebrew string to the widget. I compared the generated string between assigned to widget/not assigned to widget in the 'resources.c,' and there is indeed a small difference.
However I have a lot of strings and I don't know which widget they will be assigned to. And creating X widgets to assign my X strings to these widgets are not a viable solution

ting-ms, 2023-12-18:
@ThibaudBrandt Only strings that need line breaking should be assigned to the target widget, by doing that, GUIX Studio would know the font and the line width needed for line breaking. You may also consider disablign the option**Generate Bidi Text in Display Order** and in the application code,  reorder the BiDi text for displaying by calling API **gx_utility_bidi_paragraph_reorder**.

ting-ms, 2024-01-03:
@ThibaudBrandt Please feel free to reopen the issue if you encounter any further issues.

closed 2024-01-03 by ting-ms