eclipse-threadx/guix #122: Vertical Scroll bar And Scroll wheel

manthan1912-12, 2024-12-31:
hello Team,
I'm using **Azure GUIX Studio** with the RA6M3 MCU board, and I am using the E2 Studio IDE for development.

**_Issue 1:-_**   _Scroll Wheel  strings......_

I implemented a scroll wheel in my project to represent the menu. After setting all the strings in the scroll wheel, I built and debugged the code. While I can see the scroll wheel on my display, the strings are not visible. I have attached a snippet below for reference. Additionally, I referred to the tutorials from the GUIX SDK files but still couldn't achieve the desired result.
Here you go,
[GUIX.zip](https://github.com/user-attachments/files/18280580/GUIX.zip)

**_Issue 2:-_**   _Vertical Scroll Bar......_  

I am using a vertical scroll wheel to change data upwards or downwards using a keypad (modifying the prompt value at runtime). In Azure GUIX Studio, I added a vertical scroll bar to my screen and configured it properly. In E2 Studio, I attempted to locate the scroll bar using the **gx_widget_find** API. However, when I built and debugged the code, I couldn't find the scroll bar. Additionally, I referred to the tutorials in the GUIX SDK files but still couldn't achieve the desired result.
[image]

If anyone has a solution to my issue, please let me know as soon as possible!

Thank You in Advance!

fdesbiens, 2025-01-15:
Hi @manthan1912-12. My apologies for the delayed response.

@eclipse-threadx/iot-threadx-committers: can anyone can comment on this issue?

fdesbiens, 2025-01-28:
@eclipse-threadx/iot-threadx-committers and @eclipse-threadx/iot-threadx-contributors:  Please provide feedback if you can.