eclipse-threadx/guix #125: GUIX Studio - Is not possible to change view language

eduardo-samtronic, 2025-03-17:
In the right bar, strings section with vertical/horizontal scrollbars active, the language arrow buttons do not work, causing the section to collapse.

We have handreds of strings and 3 languages in this project. I migrated from 5.4.x version that was working correctly.

In 6.4.X version is not possible to change de language. This is important to preview the text in the screen in different languages before exporting it.

fdesbiens, 2025-03-18:
Hi @eduardo-samtronic. Thank you for opening this issue.

I will try to recreate the problem on my side. Can you share the following details about your setup, please?

- Full version number for GUIX Studio
- Windows version (as shown in the example below for my own machine)

> Edition	Windows 11 Pro
> Version	24H2
> OS build	26100.3476
> Experience	Windows Feature Experience Pack 1000.26100.54.0

If you are not using the latest 6.4.2.0 version, please upgrade and report whether the problem shows up in that one or not. You can download the installer here:
[https://github.com/eclipse-threadx/guix/releases/download/v6.4.2_rel/guix_studio_setup_version_6.4.2.0.zip](https://github.com/eclipse-threadx/guix/releases/download/v6.4.2_rel/guix_studio_setup_version_6.4.2.0.zip)

eduardo-samtronic, 2025-03-18:
Hi @fdesbiens. You're welcome.

I'm using 6.4.2.461 version, I think is the latest.

Windows version is:
Windows 11 Pro
24H2
‎06/‎01/‎2025
26100.3476

The arrow buttons I'm referring to are these:

[image]

Note that I have to scroll the horizontal bar quite a bit to see them. In version 5 they were aligned with the screen, so there was no need to scroll.

fdesbiens, 2025-03-19:
Hi @eduardo-samtronic.

I tried to reproduce the behaviour on my machine, which is running the same windows version as yourself.

I opened the "GUIX Calculator" sample. Then, I selected one of the buttons and tried to use the arrows to add a French version of the button's string. This did not work. The arrows were unresponsive as you describe above.

I then opened the `Configure` menu and selected the `Languages` option. This showed that the sample application had been configured with English as its only language. 

Using the

fdesbiens, 2025-03-21:
Hi @eduardo-samtronic.

Thank you for the clarifications. I was able to reproduce the behaviour you described.  I will attempt to fix it eventually, but I lack the time right now. 

I am due to work on GUIX Studio to implement proper scaling on high-DPI screens and could fix this issue at the same time. 

In the meantime, here is a workaround. If you double-click on the string "Calculator" in the screenshot you provided, this will bring up the "String Table Editor."

[image]

Click on the left-pointing arrow shown below.

[image]

This will provide you access to the other languages defined in the application. You can then edit the text of the string in any of the foreign languages.

[image]